react-native-camera/react-native-camera #3012: IOS recording file size is very large compare to Android

jack-solutions, 2020-10-29:
# Question

**To Do First**
- [ ] Take a look in the [README](https://github.com/react-native-community/react-native-camera/blob/master/README.md)
- [ ] Take a look in the [docs](https://github.com/react-native-community/react-native-camera/blob/master/docs/RNCamera.md)
- [ ] Take a look in the [QA](https://github.com/react-native-community/react-native-camera/blob/master/docs/QA.md)

**Ask your Question**
IOS video recording is too large 

**Tags**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

ziiw, 2021-01-19:
I have the same issue. For a 2.77GB file, I will have :
- iOS: 2775040922 RNFS.stat.size
- Android : 806681600 RNFS.stat.size

RNFS version 2.16.6

MarcoScabbiolo, 2021-01-19:
Please [ffprobe](https://ffmpeg.org/ffprobe.html) both videos and share the results so we can understand the reason behind the difference in size

closed 2021-03-27 by stale[bot]