react-native-camera/react-native-camera #3013: react-native link react-native-camera failed

ksielyov, 2020-10-30:
>react-native link react-native-camera

warn Calling react-native link [packageName] is deprecated in favor of autolinking. It will be removed in the next major release.
Autolinking documentation: https://github.com/react-native-community/cli/blob/master/docs/autolinking.md
info Linking "react-native-camera" iOS dependency
error Linking "react-native-camera" failed. Run CLI with --verbose flag for more details.

Installed the latest version of react-native-camera

ksielyov, 2020-10-30:
Could not determine the dependencies of task ':app:compileDebugJavaWithJavac'.
> Could not resolve all task dependencies for configuration ':app:debugCompileClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app

boychik-serzh, 2020-12-15:
I have the same issue. But I can't a solution, too.

closed 2021-02-20 by stale[bot]