react-native-camera/react-native-camera #3024: takePictureAsync doesn't fired, if taking photo in dark room

nfort, 2020-11-17:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 10
- iOS:
- react-native-camera: 3.40.0
- react-native: 0.62.2
- react: 16.11.0
- react-navigation: 4.4.0

**Description/Current Behaviour**

After take photo with flash in a dark room, we don't get photo or get photo with black

**Expected Behaviour**

**Steps to Reproduce**

First you need find a dark room (without light)

1. Set flash mode to auto or on
2. Press button "Take photo"
3. After about second, fired Flash and after about second Flash fired more

Sometimes takePictureAsync fired, but we get photo with only black background 

**Additionals**

Reproduced on Samsung Galaxy A40

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

closed 2021-01-23 by stale[bot]

SSISPackage, 2022-11-17:
I am facing the same issue with the react-native-camera on the Samsung A40 device. Have you found any solution for it?

nfort, 2022-11-17:
> I am facing the same issue with the react-native-camera on the Samsung A40 device. Have you found any solution for it?

sure, we are suggest user don't take photo with flash)