react-native-camera/react-native-camera #3027: [v4] Big re-writing plan 🤯🛠

renatobenks, 2020-11-19:
# Feature-Request

**Describe the Feature**

In order to keep supporting this package in a sustainable way, which it's not anymore. We'll gonna have to re-write that, to make the react-native-camera a plug-and-play package, offering to community ways to communicate with the device's camera in react-native apps without having to be modified or "evolved", to support new requirements of client's use-cases, which was the main purpose of turning it, unsustainable to be any longer maintained.

**Possible Implementations**

Splitting up react-native-camera into packages, to connect them as extensions to a single core of the camera, uncoupling almost all functionalities supported today, to work "alone". Then, the camera uses would look like:

```ts
<Camera>
  <FaceRecognition />
</Camera>
```

ps.: this "API" 👆🏻, is just a sketch of one idea, but it's something to be based on

This whole idea came from https://reactnavigation.org/blog/2020/02/06/react-navigation-5.0 and what has been done around it since then. Inspired by them, we could make the camera breathe one more time to live a long life 🎉 

**Related Issues**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

#1029

pzoltowski, 2020-12-10:
I'm not ReactNative developer (doing mostly native iOS) but I have been looking forward to having a really customizable and cross-platform Camera API. I'm just wondering if instead of reinventing the wheel wouldn't it be better to base core implementation on OpenCV. OpenCV is already cross platform, modular and suitable for realtime processing with low latency. 

Even if OpenCV is an extensive library and most react-native-camera users don't need 90% functionalities, opencv can be build just with minimum opencv/core+opencv/videoio modules. OpenCV already provides javascript bindings however those are probably compiled from C++ to WebAssembly. In case of react-native with new C++ TurboModules all we would need to have is to make RN bindings for **cv::Mat and cv::VideoCapture**. In case someone requires more advanced functionality like facedetector or ML functionality opencv already provide modules for that - again all would be needed is to expose react-native bindings for that separate modules. This would work in similar style to current opencv bindings for python, java, swift, javascript, julia. OpenCV already facilitate automatically generating bindings for those languages.

This would increase the community and instead of react-native, expo, flutter, unity, openFrameworks, unreal engine community keep reimplemmenting their own cross platform Camera API, we would have one cross-platform Camera API in C++ that anyone can contribute to with bug fixes etc., and with bindings to different languages/frameworks

Few things that I was missing from any expo, reactnative, flutter, unity, unreal engine cross platform Camera API:

- iterating available cameras and their supported formats
- choosing custom ISO and exposure time, focus, whitebalance
- choosing custom frame rate at high speed (e.g. 120fps)
- access to depth camera streams
- c++ hooks/callbacks for processing video streams in realtime.

Even though some of those features are still missing in opencv I would probably more willing to contribute at least some features for iOS based backend.

MarcoScabbiolo, 2021-01-11:
Is there any public roadmap for this? I'm ready to dedicate some time to this package.
Moving to Kotlin/Swift as suggested in #1029 sounds really good, Objective-C and Java (for Android) are being deprecated.

erikb76, 2021-01-12:
> Objective-C and Java (for Android) are being deprecated.

What do you mean by "deprecated"? Any official source for this?

MarcoScabbiolo, 2021-01-12:
> What do you mean by "deprecated"? Any official source for this?
I mean that Swift will replace Objective-C eventually and Apple is pushing in this direction. SwiftUI is a clear example of the trend.

MarcoScabbiolo, 2021-01-14:
@renatobenks Should we start discussing v4 in more detail to shape it and make it easier for those who are willing to contribute to have a clear roadmap to get coding? Maybe using a more fluid channel to communicate like [Spectrum](https://spectrum.chat/) or [Gitter](https://gitter.im/) could help.

MateusAndrade, 2021-01-14:
> @renatobenks Should we start discussing v4 in more detail to shape it and make it easier for those who are willing to contribute to have a clear roadmap to get coding? Maybe using a more fluid channel to communicate like [Spectrum](https://spectrum.chat/) or [Gitter](https://gitter.im/) could help.

We can use Gihub to manage our roadmap/project, what do you think?

MarcoScabbiolo, 2021-02-11:
For more information please head to the react-native-camera-v4 repository under this org.

closed 2021-02-11 by MarcoScabbiolo