react-native-camera/react-native-camera #3029: Cannot Run assembleAndroidTest Task

AlexK777, 2020-11-21:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: API 29
- react-native-camera: 3.40.0 (latest)
- react-native: 0.63.3
- react: 16.13.1

**Description/Current Behaviour**
We have recently updated Firebase and our dependencies to: 
```
    "@react-native-firebase/analytics": "10.0.0",
    "@react-native-firebase/app": "10.0.0",
    "@react-native-firebase/crashlytics": "10.0.0",
    "@react-native-firebase/database": "10.0.0",
    "@react-native-firebase/firestore": "10.0.0",
    "@react-native-firebase/messaging": "10.0.0",
    "@react-native-firebase/perf": "10.0.0",
    "@react-native-firebase/remote-config": "10.0.1",
```

After this update we can create debug and release builds (`assembleDebug` and `assembleRelease`), but are unable to run the `assembleAndroidTest` task, which is necessary for our end to end tests using [Detox](https://github.com/wix/Detox). We had this same issue with [another library](https://github.com/bringg/ReactNativeBringgDriverSDK) a few months ago and they were able to fix it after we reported this. 

```
> Task :react-native-camera:mergeDexMlkitDebugAndroidTest FAILED
D8: Cannot fit requested classes in a single dex file (# methods: 68308 > 65536)
com.android.builder.dexing.DexArchiveMergerException: Error while merging dex archives:
The number of method references in a .dex file cannot exceed 64K.
```

**Expected Behaviour**

The gradle task should complete successfully. 

**Steps to Reproduce**
Have react native app with latest firebase dependencies and react-native-camera. Run `./gradlew assembleAndroidTest` task.

closed 2021-01-06 by MateusAndrade

n1ru4l, 2021-01-06:
:tada: This issue has been resolved in version 3.40.5 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera/v/3.40.5)
- [GitHub release](https://github.com/react-native-camera/react-native-camera/releases/tag/v3.40.5)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

MateusAndrade, 2021-01-06:
@AlexK777 would you mind testing if this is working with the latest release?

Hatko, 2021-05-22:
I get the same issue using `"detox": "18.13.0",`

commit c2bb6ba1 (2021-01-06): fix: Enable multidexing to fix #3029 (#3030)