react-native-camera/react-native-camera #3031: firebase version 10.0.0 notworking!!

methineeinkan, 2020-11-21:
# Bug Report

**Versions**
- Android:
"react-native-camera": "^3.40.0",
"react": "16.13.1",
"react-native": "0.63.3",
 
Duplicate class com.google.android.gms.internal.vision.zzbl found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar 

[image]

Hilbertangers, 2020-11-23:
try
[image]

methineeinkan, 2020-11-23:
> try
> [image]

not work

methineeinkan, 2020-11-23:
@Hilbertangers Not Work

methineeinkan, 2020-11-23:
is work for me Go to react-native-camera => build-gradle => dependencies

 def googlePlayServicesVisionVersion = safeExtGet('googlePlayServicesVisionVersion', safeExtGet('googlePlayServicesVersion', '20.1.2'))

  //noinspection GradleDynamicVersion
  implementation 'com.facebook.react:react-native:+'  // From node_modules
  implementation "com.google.zxing:core:3.3.3"
  implementation "com.drewnoakes:metadata-extractor:2.11.0"
  generalImplementation "com.google.android.gms:play-services-vision:$googlePlayServicesVisionVersion"
  implementation "androidx.exifinterface:exifinterface:1.0.0"
  implementation "androidx.annotation:annotation:1.0.0"
  implementation "androidx.legacy:legacy-support-v4:1.0.0"
  mlkitImplementation "com.google.firebase:firebase-ml-vision:${safeExtGet('firebase-ml-vision', '24.0.3')}"
  mlkitImplementation "com.google.firebase:firebase-ml-vision-face-model:${safeExtGet('firebase-ml-vision-face-model', '20.0.1')}"

  implementation 'com.google.android.gms:play-services-vision-common:19.1.1'
  implementation "com.google.android.gms:play-services-vision-face-contour-internal:16.0.2"
  implementation 'com.google.android.gms:play-services-vision-image-label:18.0.5'

closed 2020-11-23 by methineeinkan

SnaiNeR, 2020-11-25:
same for the iOS 
[image]

methineeinkan, 2020-11-26:
> same for the iOS
> [image]

try remove podfile.look and 
rm -rf ~/Library/Caches/CocoaPods
rm -rf Pods
rm -rf ~/Library/Developer/Xcode/DerivedData/*
pod deintegrate
pod setup

SnaiNeR, 2020-11-26:
@methineeinkan it is not helped :-(
there is really conflict of dependencies, in android you resolve it by installing main version via gradle, in iOS this won't work

SnaiNeR, 2020-11-26:
```
    implementation(project(':react-native-camera'),  {
        exclude group: "com.google.android.gms"
        exclude group: "com.google.firebase", module: "firebase-ml-vision-face-model"
    })
```
this changes in app/build.gradle also helped, without editing react-native-camera package

DiMaNacho, 2020-11-27:
> same for the iOS
> [image]

Same!

yvza, 2020-11-30:
> exclude group: "com.google.android.gms"
>         exclude group: "com.google.firebase", module: "firebase-ml-vision-face-model"

Hi @SnaiNeR , i still got this error:
 Duplicate class com.google.android.gms.internal.vision.zze found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)

can u help me

rakicluca, 2021-01-04:
> def googlePlayServicesVisionVersion = safeExtGet('googlePlayServicesVisionVersion', safeExtGet('googlePlayServicesVersion', '20.1.2'))

Where is this path??? Can I update using che "Project Structure" function in Android Studio?