react-native-camera/react-native-camera #3032: CI

SimonErm, 2020-11-23:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

MateusAndrade, 2020-11-23:
Hi @SimonErm, because of a change in the organization that this repo was, probably the CI set-up is missing an Org/Plan set-tup at CircleCI. I can have a look at this, but maybe we will need some help with the set-up at CircleCI.

SimonErm, 2020-11-24:
I never worked with circleci before, but i can try to help.

SimonErm, 2020-12-28:
Any news on this or something i can do to help?
Would be great if we could merge some of the PRs soon.

MateusAndrade, 2021-01-06:
> Any news on this or something i can do to help?
> Would be great if we could merge some of the PRs soon.

Hi @SimonErm , I was able to set the project at CircleCI again. As far as I know, I think everything should be working(we still need to test the deploy to npm, etc). You can look at the build [here](https://app.circleci.com/pipelines/github/react-native-camera/react-native-camera).

closed 2021-01-06 by MateusAndrade