react-native-camera/react-native-camera #3035: Picture too dark

SirKadogan, 2020-11-25:
# Bug Report

**To Do First**

- [ X ] Did you try latest release?
- [] Did you try master?
- [ X ] Did you look for existing matching issues?

**Platforms**

This app is android only. And the issue has happened on a tablet model m73g plus.

**Versions**

ment in the optionals if used-->

- Android:
- react-native-camera: 3.40.0
- react-native: 0.60.0
- react: 16.8.6
- react-navigaton: 3.13.0

**Description/Current Behaviour**

Picture is too dark both on preview and on result. Opening the camera app is just fine, the dark result only happens on my app. Dark enough to not be able to even see the user at times.

**Expected Behaviour**

Expect to have a clear picture of my user's face.

**Steps to Reproduce**

I havent been able to reproduce the error. My client just opens the app on a M73G+ tablet and takes a picture. 

My first thought was that auto exposure was not working well, so I set a fixed 0.5 exposure and still no luck.

**Additionals**
I realize the user is taking a picture against the light, but the tablet is fixed on the wall and he can't move it. Nevertheless, camera app works just fine at the very same position.

Samples: 

[image]
[image]

tewarig, 2021-01-02:
maybe changing white balance may help

SirKadogan, 2021-01-08:
I'll give that a try

closed 2021-03-20 by stale[bot]