react-native-camera/react-native-camera #3036: NO MATTER WHAT I TRY - takePictureAsync is not returning the data

digitlninja, 2020-11-26:
# Bug Report

**To Do First**

- [ ] Did you try latest release? yes
- [ ] Did you try master? yes
- [ ] Did you look for existing matching issues? yes

**Platforms**

Nothing I do can get this console log to execute after. I've tried this with different component examples from your docs, and also making it a functiona component. The picture does capture, but the log doesn't execute, and I need the URI / to show the photo in a div in the app.

**Versions**

**Expected Behaviour**
The log executes and I can inspect the data of the photo.

**Steps to Reproduce**
Use this code
`class TakePicture extends Component {
    takePicture = async () => {
        try {
            const data = await this.camera.takePictureAsync();
            console.log('Path to image: ' + data.uri);
        } catch (err) {
            console.log('err: ', err);
        }
    };

    render() {
        return (
            <View style={styles.container}>
                <RNCamera
                    ref={cam => {
                        this.camera = cam;
                    }}
                    style={styles.preview}
                    captureAudio={false}
                >
                    <View style={{ flex: 0, flexDirection: 'row', justifyContent: 'center' }}>
                        <TouchableOpacity onPress={this.takePicture} style={styles.capture}>
                            <Text style={{ fontSize: 14 }}> Take Photo </Text>
                        </TouchableOpacity>
                    </View>
                </RNCamera>

                <View style={styles.space}/>
            </View>
        );
    }
}`

lordofflakes, 2020-11-27:
If you're testing on android please check that your app has storage write permissions.

digitlninja, 2020-11-27:
How?
I put these in AndroidManifest
` <uses-permission android:name="android.permission.CAMERA"/>

    <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE"/>
    <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>`

lordofflakes, 2020-11-30:
Use PermissionsAndroid (https://reactnative.dev/docs/permissionsandroid ) to ask user for WRITE_EXTERNAL_STORAGE permissions.

davidkdev, 2021-01-05:
I have the same problem, did you find a solution? @digitlninja

closed 2021-03-13 by stale[bot]