react-native-camera/react-native-camera #3040: Can't run on iOS after update firebase to 10.x.x

DiMaNacho, 2020-11-30:
# Bug Report

**To Do First**

- [ ☑️ ] Did you try latest release?
- [ ☑️ ] Did you try master?
- [ ☑️ ] Did you look for existing matching issues?

**Versions**

```json
    "react-native-camera": "^3.40.0",
    "react-native": "0.63.3",
    "react": "16.13.1",
    "@react-native-firebase/analytics": "^10.1.0",
    "@react-native-firebase/app": "^10.1.0",
    "@react-native-firebase/crashlytics": "^10.1.0",
    "@react-native-firebase/messaging": "^10.1.0",
    "@react-native-firebase/remote-config": "^10.1.0",
```

**Description/Current Behaviour**

After update firebase dependencies i can't did `pod install`.

- I removed `podfile.lock`
- I did `pod update`
- I reinstalled all dependencies.
- I cleaned `node_modules` (`rm -rf node_modules`)
- I cleaned  npm cache (`npm cache clean --force`)
- I cleaned yarn cache (`yarn cache clean`)
- And I cleaned iOS folder (`rm -rf Pods && rm -fr DerivedData && pod cache clean --all && rm -fr build && pod install --repo-update`).

A lot of times 😅

[image]

DiMaNacho, 2020-11-30:
This PR:

https://github.com/react-native-camera/react-native-camera/pull/2931

Solve my problem.

ferasabufares, 2020-12-02:
can you help me please i am stuck here how did you solve this problem 

and did onGoogleVisionBarcodesDetected work on ios ??

please help me

DiMaNacho, 2020-12-02:
> can you help me please i am stuck here how did you solve this problem
> 
> and did onGoogleVisionBarcodesDetected work on ios ??
> 
> please help me

@ferasabufares did you update firebase libreries? About the [PR] (https://github.com/react-native-camera/react-native-camera/pull/2931) i did `yarn add davidgovea / react-native-camera # standalone-mlkit` temporally, waiting their merging.

Gustavo-Kuze, 2020-12-08:
@DiMaNacho Did you managed to get it working on Android? I'm trying to reference `davidgovea/react-native-camera#standalone-mlkit` in my package.json as you mentioned, but I'm getting the following error when I try to execute `npx react-native run-android`:

```
\react-native-camera\android\src\main\java\org\reactnative\camera\RNCameraViewHelper.java:22: error: package org.reactnative.imagelabeler does not exist
```

DiMaNacho, 2020-12-08:
> @DiMaNacho Did you managed to get this working on Android? I'm trying to reference `davidgovea/react-native-camera#standalone-mlkit` in my package.json as you mentioned, but I'm getting the following error when I try to execute `npx react-native run-android`:
> 
> ```
> \react-native-camera\android\src\main\java\org\reactnative\camera\RNCameraViewHelper.java:22: error: package org.reactnative.imagelabeler does not exist
> ```

yes I did, maybe you need to fix something on Android folder, try to read react-native-firebase documentation.

DiMaNacho, 2020-12-08:
@Gustavo-Kuze

current package.json

```json
    "@react-native-firebase/analytics": "^10.1.0",
    "@react-native-firebase/app": "^10.1.0",
    "@react-native-firebase/crashlytics": "^10.1.0",
    "@react-native-firebase/messaging": "^10.1.0",
    "@react-native-firebase/remote-config": "^10.1.0",
    "react-native-camera": "davidgovea/react-native-camera#standalone-mlkit",
```

Changes I made on `android/app/build.gradle`

from 
```gradle
implementation 'com.android.support:multidex:1.0.3'
```
to 
```gradle
    implementation 'androidx.multidex:multidex:2.0.1'
```

and 
```gradle
implementation project(':react-native-camera')`
```

to 
```gradle
implementation (project(':react-native-camera')) {
        exclude group: "com.google.android.gms"
        exclude group: "com.google.android.gms", module: "play-services-vision"
        exclude group: "com.google.firebase", module: "firebase-ml-vision-face-model"
    }

    implementation platform('com.google.firebase:firebase-bom:26.1.0')
    implementation 'com.google.firebase:firebase-analytics'
```

and thats all I made for android.

saczuac, 2020-12-09:
Same issue here

yudiedrialviska, 2020-12-17:
same issue for me

DerrickC, 2021-01-26:
Same issue!!

99uno, 2021-02-09:
Same issue!

tanmoythander, 2021-03-25:
Same issue!

Shahulul, 2021-03-26:
Same issue

saczuac, 2021-03-31:
Can someone fix RNCamera PLZ ????

gabrieljoelc, 2021-04-11:
I was trying to upgrade `@react-native-firebase/analytics`:
- Upgrade to `11.2.0`, uses Firebase/Core `7.9.0`
- RNCamera `3.43.5`, uses Firebase/Core `5.0.0`

I had to downgrade all ``@react-native-firebase/*` packages to `8.x.x` to get pods to install. :(

GunnarAK, 2021-04-23:
Still working out the barcode recognition (Android OK, iOS unclear) but got it working with the following setup:

package.json
```
dependencies: {
    "@react-native-firebase/analytics": "^10.8.1",
    "@react-native-firebase/app": "^10.8.1",
    "@react-native-firebase/crashlytics": "^10.8.1",
    "@react-native-firebase/messaging": "^10.8.1",
    "@react-native-firebase/remote-config": "^10.8.1",
    ...
    "react-native-camera": "davidgovea/react-native-camera#standalone-mlkit",
}
```
Podfile
```
pod "react-native-camera", path: "../node_modules/react-native-camera", subspecs: [
                               "BarcodeDetectorMLKit",
                             ]
```
android/app/build.gradle
```
  defaultConfig {
    ...
    missingDimensionStrategy 'react-native-camera', 'mlkit'
  }

dependencies {
   ...
   implementation "com.google.android.gms:play-services-base:17.6.0"
   ...
}
```
AndroidManifest.xml
```
<application
   ...
   >
   <meta-data
         android:name="com.google.firebase.ml.vision.DEPENDENCIES"
         android:value="ocr, barcode" />
...
</application>
```

saczuac, 2021-04-26:
@n1ru4l

n1ru4l, 2021-04-26:
@saczuac please stop spam tagging me. Try to implement a fix and create a pull request. I am not actively maintaining this project right now.

chiubaka, 2021-05-13:
I was blocked on trying to get `react-native-camera` to operate side-by-side with `@react-native-firebase/analytics` and stumbled upon this issue. I tried a bunch of other things like the davidgovea branch. Some of those partially fixed the issue, but I encountered other problems and didn't feel good about being on a closed branch that isn't getting library updates.

Ultimately, I was able to get things resolved by downgrading my `@react-native-firebase/analytics` and `@react-native-firebase/app` versions to something compatible with `react-native-camera`. I specifically have:

```
{
  ...
  "@react-native-firebase/analytics": "^7.6.10",
  "@react-native-firebase/app": "^8.4.7",
  "react-native-camera": "^3.43.6",
  ...
}
```

I later discovered that this configuration was actually causing `react-native-camera` to crash when the camera started on Android. I tracked that problem down to a dependency conflict where some of the `firebase-ml` dependencies for Android were getting bump up several major versions because of the Firebase dependency versions included by `@react-native-firebase/app` even with my lowered npm package versions.

It's not pretty, but I was able to resolve that with the following in my app's `build.gradle`:

```
    implementation (project(":react-native-camera")) {
        exclude group: "com.google.android.gms", module: "play-services-vision"
        exclude group: "com.google.android.gms", module: "play-services-vision-common"
        exclude group: "com.google.android.gms", module: "play-services-vision-face-contour-internal"
        implementation("com.google.firebase:firebase-ml-common") {
            version {
                strictly "17.0.0"
            }
        }
        implementation("com.google.firebase:firebase-ml-vision") {
            version {
                strictly "19.0.3"
            }
        }
        implementation("com.google.firebase:firebase-ml-vision-face-model") {
            version {
                strictly "17.0.2"
            }
        }
    }
```

Just wanted to leave this somewhere so others can benefit from the hours I just spent learning way too much about gradle and searching through dependency scans to figure out what was going wrong.

In actuality, I think the long-term solution for my project is to deprecate `react-native-camera` in favor of `react-native-vision-camera` since this repository doesn't seem maintained :/.

GunnarAK, 2021-05-14:
@Gustavo-Kuze how did you fix your issues with the image labeler? I'm getting the same issue and reading the RNFirebase docs did not help me any further unfortunately.
The error occurs only during release builds.

build error: 
```
node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/events/LabelDetectionErrorEvent.java:11: 
error: package org.reactnative.imagelabeler does not exist
```
I'm running @davidgovea's fork which builds&runs perfectly fine for debug builds.

Gustavo-Kuze, 2021-05-14:
@GunnarAK 
Unfortunately I didn't fix the issues for this library 😔. I ended up replacing `react-native-camera` with [expo-camera](https://github.com/expo/expo/tree/master/packages/expo-camera) using [react-native-unimodules](https://docs.expo.io/bare/installing-unimodules/), since the project is "bare react-native app".

GunnarAK, 2021-05-14:
Ha I just managed to fix this by hopelessly trying stuff. Came across the following issue https://github.com/wix/react-native-navigation/issues/4756#issuecomment-465199365
This proposed the following:
```
./gradlew app:assembleRelease
```

That fixed my issue. From what I understand gradle could not pick the correct module to execute `assembleRelease` on.

HyopeR, 2021-05-28:
Any improvement?

mja0342, 2021-07-09:
Why we are not able to use upgraded firebase with react-native-camera?

willsr, 2021-08-15:
> Ha I just managed to fix this by hopelessly trying stuff. Came across the following issue [wix/react-native-navigation#4756 (comment)](https://github.com/wix/react-native-navigation/issues/4756#issuecomment-465199365)
> This proposed the following:
> 
> ```
> ./gradlew app:assembleRelease
> ```
> 
> That fixed my issue. From what I understand gradle could not pick the correct module to execute `assembleRelease` on.

If you could distill the react native experience into a sentence, this would be it.