react-native-camera/react-native-camera #3043: Unable to find a specification for `React-Core` depended upon by `react-native-camera/BarcodeDetectorMLKit`

tcagkansokmen, 2020-12-02:
```
"dependencies": {
    "@codler/react-native-keyboard-aware-scroll-view": "^1.0.1",
    "axios": "^0.19.0",
    "entities": "^2.1.0",
    "expo": "^38.0.8",
    "expo-cli": "^3.28.5",
    "native-base": "github:Healthyco/NativeBase#feature/fix-request-animation",
    "native-base-autocomplete": "^1.3.2",
    "prop-types": "^15.7.2",
    "react": "16.9.0",
    "react-native": "0.62.0",
    "react-native-camera": "^3.40.0",
    "react-native-camera-hooks": "^0.3.1",
    "react-native-chart-kit": "^6.7.0",
    "react-native-daterange-picker": "^1.4.0",
    "react-native-gesture-handler": "^1.3.0",
    "react-native-keyboard-aware-scroll-view": "https://github.com/APSL/react-native-keyboard-aware-scroll-view#fb40649a9c84aa23d6a55355e9e0e432795967a6",
    "react-native-keyboard-aware-scrollview": "^2.1.0",
    "react-native-linear-gradient": "^2.5.6",
    "react-native-masked-text": "^1.13.0",
    "react-native-modal": "^11.5.6",
    "react-native-safe-area-context": "~3.0.7",
    "react-native-screens": "^2.7.0",
    "react-native-svg": "^12.1.0",
    "react-native-swipe-list-view": "^3.2.5",
    "react-native-vector-icons": "^7.1.0",
    "react-navigation": "^3.11.2",
    "react-redux": "^7.2.0",
    "redux": "^4.0.1",
    "redux-persist": "^5.10.0",
    "redux-saga": "^1.0.2",
    "reduxsauce": "^1.0.1",
    "victory-native": "^35.3.1"
  },
```

I'm getting this error when I type 'pod install" after I add "pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
    'BarcodeDetectorMLKit'
  ]" line to podfile.

Unable to find a specification for `React-Core` depended upon by `react-native-camera/BarcodeDetectorMLKit`

You have either:
 * out-of-date source repos which you can update with `pod repo update` or with `pod install --repo-update`.
 * mistyped the name or version.
 * not added the source repo that hosts the Podspec to your Podfile.

axeljeremy7, 2021-01-19:
any update on this? I am having the same issue. I think is because is not compatible with 0.63.4, I am upgrading from 0.61.5

99uno, 2021-02-08:
Same issue, any idea?

99uno, 2021-02-10:
I found the problem: must rename dependencies in react-native-camera.podspec. I did it manually until someona can update this library

[image]

[image]

rukmanary, 2021-03-05:
> I found the problem: must rename dependencies in react-native-camera.podspec. I did it manually until someone can update this library
> 
> [image]
> 
> [image]

This is works for me. Thanks

closed 2021-05-12 by stale[bot]