react-native-camera/react-native-camera #3050: Device doesn't support flash mode

Romick2005, 2020-12-15:
I got warning every time I use camera on my old iPhone 5s. 
Flash is not disabled and works properly with camera.

On earlier versions of react-native-camera everything was working as expected.

[RNCamera updateFlashMode]: device doesn't support flash mode

hamzaouhssain1993, 2020-12-22:
I do not understand. There is a method inside the project that is "hasFlassh()" but we are not able to use it. and for that we are not able to disable the flash if the camera does not support it at all.

caojianfeng, 2020-12-24:
Same issues on ipad mini4 (ios 13.5.1)

gyoung-panopto, 2021-01-29:
I'm hitting this issue on an iPad as well. Fortunately it just seems to warn and doesn't affect the behavior, but the log spam is annoying.

Romick2005, 2021-01-30:
> I do not understand. There is a method inside the project that is hasFlash() but we are not able to use it. and for that we are not able to disable the flash if the camera does not support it at all.

Device camera actually support flash mode, but the library give a warning that saying the opposite.
Tested iOS 12.5.1

closed 2021-04-07 by stale[bot]

tsoisauce, 2021-04-13:
Having the same issue, creating an internal business tool for iPad that reads barcodes and I am unable to exclude `flashMode`

Any help would be kindly appreciated.

Romick2005, 2021-04-14:
#3183 I removed this annoying unneeded warning in this pull request.
AS for me initially I assumed that device = iPhone and I was using iPhone 5s that has two camera (back camera has flash mode and front camera does not have flash mode). So I was thinking that this message is wrong as my device support flash mode. But after short investigation I discovered that device is not an iPhone itself it is only camera device so than when I was using front camera and in the app I use only front camera, I got this warning, but it doesn't tell anything useful or something that I can fix so to me it seems that a better solution for this is just to remove this warning as code logic does not brake and works as expected.

Romick2005, 2021-05-12:
Until those warning are removed I use:
`
LogBox.ignoreLogs(
  [
    "[RNCamera updateFlashMode]: device doesn't support flash mode"
  ]
);
`