react-native-camera/react-native-camera #3053: onTextRecognized() method doesn't call in IOS

saurabh874, 2020-12-18:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

tungduonghgg123, 2021-01-07:
In order to make `onTextRecognized` method running on iOS, it is required to manually install additional dependencies, which is FirebaseMLVision through pods. You should check the guide in here, the `Additional installation steps` section. 
[Guide](https://github.com/react-native-camera/react-native-camera/blob/master/docs/installation.md#requirements)
I face the same problem lately and this fixes it. Hopefully it helps :)

saurabh874, 2021-01-07:
@tungduonghgg123  if i am doing same adding pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
  'TextDetector'
] in podFile then run pod install or pod update i am getting firebase related error.
**[!] CocoaPods could not find compatible versions for pod "Firebase/CoreOnly":**
**react-native-camera/TextDetector (from `../node_modules/react-native-camera`) was resolved to 3.40.0, which depends on
      Firebase/MLVisionTextModel was resolved to 5.0.0, which depends on
        Firebase/CoreOnly (= 5.0.0)**

MateusAndrade, 2021-01-07:
Hi, @saurabh874 is this working with any of the example?

99uno, 2021-02-09:
Same issue

closed 2021-04-18 by stale[bot]

ivankozlo, 2022-03-02:
> @tungduonghgg123 if i am doing same adding pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [ 'TextDetector' ] in podFile then run pod install or pod update i am [...]

I am having the exact same issue. For the production related problem, I had to use rect-native-camera version 3.30.0 and working on iOS15.2.