react-native-camera/react-native-camera #3056: after stopRecording the app freezes in android

marquina-abreu, 2020-12-20:
# Bug Report
 After stopRecording the app freezes in android
**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**
android

**Versions**
react-native: "0.63.2"
react-native-camera: "3.40.0"

- Android:
- iOS:
- react-native-camera: "3.40.0"
- react-native: "0.63.2"
- react: "16.13.1"
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

fansmc, 2020-12-30:
三星SM-G9860 机器上遇到了同样的问题

Hermanyo, 2021-01-29:
Same issue, anyone solved it?

closed 2021-04-06 by stale[bot]