react-native-camera/react-native-camera #3057: Mocking onBarCodeRead event in jest

Jp3rd, 2020-12-21:
# How to mock the event to run onBarCodeRead in jest?
I am trying to write tests for the flow of scanning a QR code, is there a way to mock emitting the event for onBarCodeRead to run?

I am running on Android
"react-native": "0.61.5"
"react-native-camera": "^3.19.0"

jjairojr, 2021-01-29:
@Jp3rd You got it ? i'm having the same issue T.T

closed 2021-04-06 by stale[bot]

airballooninc, 2021-06-23:
I've solved this issue by spying on a scanner component that uses RNCamera

`// real scanner component that reads barcode data from RNCamera

const Scanner = ({onBarCodeRead}) => {
  return (
    <View accessibilityLabel="scanner">
      <RNCamera
        accessibilityLabel="camera"
        style={styles.camera}
        onBarCodeRead={({data}) => onBarCodeRead(data)}
      />
    </View>
  );
};

export default Scanner;

// spy on scanner component in the test file

import * as Scanner from './Scanner'; // import the scanner component to spy on it

const mockData = 'code*name*address'; // this is my mocked barcode data that will be returned by onBarcodeRead
 
jest.spyOn(Scanner, 'default').mockImplementation(({onBarCodeRead}) => (
  <View accessibilityLabel="scanner">
    <TouchableOpacity
      accessibilityLabel="mockOnBarCodeRead"
      onPress={() => onBarCodeRead(mockData)}
    />
  </View>
));  // triggers onBarCodeRead on press

// to trigger onBarCodeRead during test
const {getByLabelText} = render(TestComponent)

fireEvent.press(
      within(getByLabelText('scanner')).getByLabelText('mockOnBarCodeRead'),
    );`

airballooninc, 2021-06-23:
I've mocked react-native-camera module's RNCamera to trigger onBarCode read on press.

```
const mockRNCamera = (onBarCodeRead: ({data}: {data: string}) => void) => (
  <View accessibilityLabel="scanner">
    <TouchableOpacity
      accessibilityLabel="mockOnBarCodeRead"
      onPress={() => onBarCodeRead({data: mockData})}
    />
  </View>
);

jest.mock('react-native-camera', () => ({
  RNCamera: ({
    onBarCodeRead,
  }: {
    onBarCodeRead: ({data}: {data: string}) => void;
  }) => mockRNCamera(onBarCodeRead),
}));

const mockRNCamera = (onBarCodeRead: ({data}: {data: string}) => void) => (
  <View accessibilityLabel="scanner">
    <TouchableOpacity
      accessibilityLabel="mockOnBarCodeRead"
      onPress={() => onBarCodeRead({data: mockData})}
    />
  </View>
);

jest.mock('react-native-camera', () => ({
  RNCamera: ({onBarCodeRead}: {onBarCodeRead: () => void}) =>
    mockRNCamera(onBarCodeRead),
}));

```

hyan-campos, 2021-06-23:
@airballooninc How do you call the click on the button "mockOnBarCodeRead"

airballooninc, 2021-06-24:
@hyan-campos I call the on press event like so in my test file. The gist of how I setup my test.

```
import React, {useState} from 'react';
import {Text, TouchableOpacity, View} from 'react-native';
import {fireEvent, render} from '@testing-library/react-native'; // using react-native-testing-library to fire events
import {RNCamera} from 'react-native-camera';

// simple test component that uses RNCamera's onBarCodeRead.
const TestComponent = () => {
  const [data, setData] = useState<string>();
  return (
    <>
      <RNCamera
        accessibilityLabel="camera"
        onBarCodeRead={({data: newData}) => {
          console.log(
            `onBarCodeRead has been triggered! with data: ${newData}`,
            setData(newData),
          );
        }}
      />
      {data && <Text>{data}</Text>}
    </>
  );
};

// mock data to be read by onBarCodeRead
const mockData = 'code*name*address';

const mockRNCamera = (onBarCodeRead: ({data}: {data: string}) => void) => (
  <View accessibilityLabel="scanner">
    <TouchableOpacity
      accessibilityLabel="mockOnBarCodeRead"
      onPress={() => onBarCodeRead({data: mockData})}
    />
  </View>
);

jest.mock('react-native-camera', () => ({
  RNCamera: ({
    onBarCodeRead,
  }: {
    onBarCodeRead: ({data}: {data: string}) => void;
  }) => mockRNCamera(onBarCodeRead),
}));

const inst = render(<TestComponent />); // render the TestComponent, RNCamera component will be mocked and mockOnBarCodeRead will be rendered instead.

test('should call onBarCodeRead', () => {
  const {getByLabelText, getByText} = inst;

  // fire onBarCodeRead
  fireEvent.press(getByLabelText('mockOnBarCodeRead'));

  // expect data to be rendered as text
  getByText(mockData);
});

```