react-native-camera/react-native-camera #3060: Android: takePictureAsync() returns grey image

joselayana, 2020-12-21:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 16-29
- react-native-camera: 3.40.0
- react-native: 0.63.3
- react: 16.13.1
- react-navigation/core: 5.14.3

**Description/Current Behaviour**

When execute takePictureAsync() the image returns grey.

**Steps to Reproduce**

Device: samsung SM-G532M
Level: OS Level 23

[image]

timjbray, 2021-04-12:
Hi @joselayana, We have noticed an issue just like this with a customer running Samsung TAB 9 on Android 7. Did you end up resolving it or find a workaround?

RN: 0.63.4, React: 17.0.1, RN Camera: 3.42.3

marknelissen, 2021-05-18:
Same observation on a Samsung Galaxy Tab S2 running Android 7 as well.

RN: 0.63.4, React: 16.13.1, RN Camera: 3.41.0

vsheyanov, 2021-05-25:
Same problem, Samsung Galaxy Tab S2 running Android 7 (not updated for 2-3 years).

hamzaouhssain1993, 2021-05-31:
Same issue on Samsung galaxy tab S2 running Android 7

hamzaouhssain1993, 2021-07-30:
No update on this issue? Thanks

renatobenks, 2021-08-03:
I'll be trying to reproduce it on a physical device this week as I didn't get this same error in a simulator, but I think it's probably related to some change that Samsung has made on Android UI engine in a way that it's not currently supported by our drawing process of the image captured. Fortunately, I have a Galaxy Tab in hands.

hamzaouhssain1993, 2021-08-03:
Thank you a lot for your effort, Renato. I will keep you update if we find something on our side.
From the Android Log we got this:
D/skia: libjpeg error 117 Corrupt JPEG data: premature end of data segment from output_message

It might be related to the issue

marknelissen, 2021-08-04:
@renatobenks , is it de Tab S2? Since it's very model specific, other models in de Galaxy Tab line don't suffer this issue according to our tests.

drash-course, 2021-08-04:
@renatobenks We have the same problem on the Samsung Tab SM-T813.

I digged in the code and found out that Android's `Camera.PictureCallback::onPictureTaken(byte[] data, Camera camera)` is returning corrupted a corrupted file. The `byte[] data` array is full of 0x00 after a few kB of data. This happens when `mCameraParameters.setRotation(...)` is set to any rotation besides 0 on this device.

Since you already have the facilities in `ResolveTakenPictureAsyncTask.java` to handle the rotation in software, a workaround is to do `mCameraParameters.setRotation(0)` and perform the rotation in the async task. Here is a patch with my implementation. Can you take a look? I can also open a PR if it's more convenient. 

[react-native-camera+3.44.3.patch.zip](https://github.com/react-native-camera/react-native-camera/files/6930857/react-native-camera%2B3.44.3.patch.zip)

renatobenks, 2021-08-04:
That's great @drash-course, a wonderful explanation as well! So, please, open a PR for this patch, and let's make some further tests to see what we got, but it sounds consistent to solve the problem so far.

drash-course, 2021-08-05:
Ok will do. We'll need a list of affected devices though. @marknelissen can you confirm which device model you have on hand which is affected as well?

marknelissen, 2021-08-07:
Galaxy Tab S2, but I won't have access for the next 2 weeks. But maybe my colleague @wolfy07 can help you.

renatobenks, 2021-08-09:
Let us know how the fix sounded to you @marknelissen

wolfy07, 2021-08-31:
Hello just tried on Samsung S2 seems to be solved Thanks for the fix

marknelissen, 2021-11-03:
@renatobenks , this issue can be resolved, it seems.

renatobenks, 2021-11-09:
great

closed 2021-11-09 by renatobenks