react-native-camera/react-native-camera #3064: Build apk failed!

xxss0903, 2020-12-25:
When build release apk, always failed by below error!

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:lintVitalRelease'.
> Could not resolve all artifacts for configuration ':react-native-camera:mlkitDebugRuntimeClasspath'.
   > Could not resolve com.google.firebase:firebase-ml-common:[17.0.0].
     Required by:
         project :react-native-camera > com.google.firebase:firebase-ml-vision:19.0.3
      > Failed to list versions for com.google.firebase:firebase-ml-common.
         > Unable to load Maven meta-data from https://maven.google.com/com/google/firebase/firebase-ml-common/maven-metadata.xml.
            > Could not HEAD 'https://maven.google.com/com/google/firebase/firebase-ml-common/maven-metadata.xml'.
               > Connect to maven.google.com:443 [maven.google.com/216.58.200.238] failed: connect timed out

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

Deprecated Gradle features were used in this build, making it incompatible with Gradle 6.0.
Use '--warning-mode all' to show the individual deprecation warnings.

nick22985, 2020-12-29:
I have the exact same issue

tewarig, 2021-01-02:
I was also facing a similar error issue earlier.  go to **android > app > build.gradle**
add ** missingDimensionStrategy 'react-native-camera','general' ** in the defaultConfig
after which your defaultConfig will look like this.
`
    defaultConfig {
        applicationId "com.camera"
        minSdkVersion rootProject.ext.minSdkVersion
        targetSdkVersion rootProject.ext.targetSdkVersion
        versionCode 1
        versionName "1.0"
        missingDimensionStrategy 'react-native-camera','general'

    }`
Hope that helps.

closed 2021-03-20 by stale[bot]