react-native-camera/react-native-camera #3068: Firebase ML version lib class match issue

rajeev-tomar06, 2020-12-30:
# Bug Report

We are using 'https://rnfirebase.io/' for PushNotification & Analytics. After upgrading the lib we are getting the build issue on Android platform for duplicate class error. It's might be both lib camera & firebase using the firebase ML vision lib. How can we handle this situation. Can we exclude this lib -"firebase ML vision" somehow.

russelRajitha, 2021-01-01:
@rajeev-tomar06 do you have any specific versions of firebase react native and other stuff?

rajeev-tomar06, 2021-01-05:
@russelRajitha Please find the details below.
"react-native-camera": "^3.31.1",
========================
    "@react-native-firebase/analytics": "^10.3.1",
    "@react-native-firebase/app": "^10.3.0",
    "@react-native-firebase/messaging": "^10.3.1",
=======================================

Muhammad-Wasi, 2021-03-04:
> manage to fix the issue by adding following
> 
> ```
>    implementation (project(':react-native-camera')) {
>         exclude group: "com.google.android.gms"
>         exclude group: "com.google.android.gms", module: "play-services-vision"
>         exclude group: "com.google.firebase", module: "firebase-ml-vision-face-model"
>     }
> 
>     implementation ("com.google.firebase:firebase-ml-vision:19.0.3") {
>         force = true;
>     }
>     implementation ("com.google.firebase:firebase-ml-vision-face-model:17.0.2") {
>         force = true;
>     }
> ```

Worked in my case <3

tanmoythander, 2021-03-25:
> manage to fix the issue by adding following
>
> ```
>    implementation (project(':react-native-camera')) {
>         exclude group: "com.google.android.gms"
>         exclude group: [...]

Added this in **app/build.gradle** inside **dependencies** and it worked like magic !!!

closed 2021-05-05 by renatobenks