react-native-camera/react-native-camera #3076: CocoaPods could not find compatible versions for pod "Firebase/CoreOnly": with react-native-camera/textdetector

saurabh874, 2021-01-08:
Hi i am using firebase v6.
"@react-native-firebase/app": "^10.1.0",
"@react-native-firebase/crashlytics": "^8.4.9",
"@react-native-firebase/dynamic-links": "^7.5.9",
"@react-native-firebase/messaging": "^10.2.0",
"@react-native-firebase/ml": "^10.1.1",
"react-native-camera": "^3.40.0",
i need textdetector which is come in react-native-camera
pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
'TextDetector'
] so when i am adding this line in my podfile. i am getting some error which is related from firebase.

**[!] CocoaPods could not find compatible versions for pod "Firebase/CoreOnly":
In Podfile:
RNFBAnalytics (from ../node_modules/@react-native-firebase/analytics) was resolved to 10.3.0, which depends on
Firebase/Analytics (~> 7.3.0) was resolved to 7.3.0, which depends on
Firebase/Core (= 7.3.0) was resolved to 7.3.0, which depends on
Firebase/CoreOnly (= 7.3.0)

RNFBApp (from `../node_modules/@react-native-firebase/app`) was resolved to 10.3.0, which depends on
  Firebase/CoreOnly (~> 7.3.0)

react-native-camera/TextDetector (from `../node_modules/react-native-camera`) was resolved to 3.40.0, which depends on
  Firebase/MLVisionTextModel was resolved to 5.0.0, which depends on
    Firebase/CoreOnly (= 5.0.0)**
    

  anybody have idea what is problem here. i tried with pod update pod install. it seem to be mismatch version. react-native-camera still supporting  old version of firebase.

MateusAndrade, 2021-01-08:
Did you try updating the deps that rn-camera is using? (Editing directly at the node_modules)

saurabh874, 2021-01-11:
No @MateusAndrade. is it any way to resolve this one?

zappingseb, 2021-01-11:
I am having the same issue. Seems to be related to FireBaseAnalytics:

PodFile
```
pod 'Firebase/Analytics'

...

pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
    'TextDetector'
  ]
  #For BarCode Recognition:
  pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
    'BarcodeDetectorMLKit'
  ]
```

zappingseb, 2021-01-12:
I guess this can be closed, it is the same issue as #3040

hexatown-m-tayyab, 2021-01-29:
use this ... pod update Firebase/CoreOnly

Zeneixe, 2021-02-02:
This can't be closed, we have the same problem and `pod update Firebase/CoreOnly` won't fix it

n1dennis, 2021-02-08:
I was getting the same errors that magically went away by initializing a new project w/ npm instead of yarn and then following this: 
https://github.com/react-native-camera/react-native-camera/blob/master/docs/installation.md#requirements
Don't know if it will work for you but hope you all get it working.

99uno, 2021-02-09:
Same issue, any update?

tatva-hiral, 2021-02-12:
Same error

anhtuank7c, 2021-02-13:
I had an incompatible issue.

[image]

faizantariq1, 2021-02-17:
Hi @saurabh874 did anyone solve this issue?

dellert, 2021-02-19:
still have this problem too

faizantariq1, 2021-02-20:
Hi @dellert able to solve this problem.
In react native package.json replace the package name of "react-native-camera": "^3.42.2",  into  "react-native-camera": "davidgovea/react-native-camera#standalone-mlkit",   and after that in the main directory use npm install to install this. it will install standalone mlkit so you can this mlkit for TextDetector and also you can install @react-native-firebase/app library and it will be working fine.
if you still find the issue of version conflict in ios. using pod install then firstly delete the Podfile.lock then install pods.

nboclzz, 2021-02-27:
Deleting Podfile.lock then pod install solved my problem.

AlexHolubDoIt, 2021-03-01:
any update? have this issue too

rubenteste, 2021-03-01:
similar error:
if i try to add some plugin on my pusbpec.yaml and run again the project give me this error:
android so easy, IOS only problems......

Resolving dependencies of `Podfile`
      CDN: trunk Relative path: CocoaPods-version.yml exists! Returning local because checking is only perfomed in repo update
    [!] CocoaPods could not find compatible versions for pod "arkit_plugin":
      In Podfile:
        arkit_plugin (from `.symlinks/plugins/arkit_plugin/ios`)

JonathanHallen1991, 2021-03-03:
I got this issue from following the installation guide on firebase console. This does not take auto linking in to consideration for obvious reasons. So running `pod install --repo-update` as specified in the installation guide on https://rnfirebase.io/#4-autolinking--rebuilding solved the issue for me. Guess this was a 60 minute lesson in the importance of reading the entire guide and not just skimming through it...

rozd, 2021-04-19:
You may find helpful this workaround that worked for me for the same error on Flutter: https://github.com/FirebaseExtended/flutterfire/issues/4625#issuecomment-821792539

manjinderr, 2021-04-27:
Any update when this issue would be fixed.

I also see similar error:
[!] CocoaPods could not find compatible versions for pod "Firebase/CoreOnly":
In Podfile:
RNFBApp (from ../node_modules/@react-native-firebase/app) was resolved to 11.3.3, which depends on
Firebase/CoreOnly (= 7.10.0)
react-native-camera/TextDetector (from ../node_modules/react-native-camera) was resolved to 3.43.6, which depends on
Firebase/MLVisionTextModel was resolved to 5.0.0, which depends on
Firebase/CoreOnly (= 5.0.0)

Packages used:
"react-native-camera": "^3.43.6",
"@react-native-firebase/app": "^11.3.3",

lucashsilva, 2021-04-28:
I have the same problem

Alejandrehl, 2021-04-28:
I have the same problem

imran1992, 2021-05-06:
$ rm Podfile.lock
$ pod install --repo-update

nthegedus, 2021-05-13:
Any solution?

urmatt, 2021-05-16:
rm -rf Podfile.lock
pod install --repo-update

worked for me

nthegedus, 2021-05-17:
@urmatt what is your environment and packages version used?

imamabdulazis, 2021-05-18:
> ```
> rm -rf Podfile.lock
> pod install --repo-update
> ```
> 
> worked for me
Still not working for me, my case is remove old `react-native-firebase` with `@react-native-firebase/app`.
Anyone have same problem?

HyopeR, 2021-06-08:
Is there any progress towards solving the problem? same problem :/

mklb, 2021-06-09:
Repo update helped 👉🏻  `pod install --repo-update`

temchannat, 2021-06-14:
`pod install --repo-update` Help me too

HyopeR, 2021-06-14:
@mklb I tried this. However, I did not get any results. pod install does not happen because CoreOnly versions are different. Do you have a suggestion? I downgraded firebase packages as a workaround. I can use it without conflict in @react-native-firebase/app -> 8.4.7 version. But this version is 6 months in advance.

nthegedus, 2021-06-15:
@HyopeR I think the only solution is to use the version you mentioned.
Here's how is on my `package.json` using the most "recent" versions

```
"@react-native-firebase/analytics": "^10.0.0",
"@react-native-firebase/app": "^8.4.7",
"@react-native-firebase/crashlytics": "^8.5.2",
"@react-native-firebase/dynamic-links": "^7.5.13",
"@react-native-firebase/messaging": "^8.0.1",
"@react-native-firebase/remote-config": "^10.0.1",
```

mklb, 2021-06-18:
@HyopeR maybe delete the `Pods` folder first and than do the `pod install --repo-update`. Sometimes deleting the folder helps with issues like that..

nthegedus, 2021-09-02:
This issue has fixed with 4.0.0 release https://github.com/react-native-camera/react-native-camera/issues/3166#issuecomment-881571214

charles-ekube, 2022-06-17:
> ```
> rm -rf Podfile.lock
> pod install --repo-update
> ```
> 
worked for me

miguel-martinez-6661, 2022-10-24:
The only solution I've found.

```
rm -rf Podfile.lock
pod install --repo-update
```

RezaHabibi1001, 2023-01-19:
for me   running these below in app/ ios commands solved. problem :

`pod repo update`
`pod update`