react-native-camera/react-native-camera #3080: Build Failed,Could not resolve com.google.android.gms:play-services-vision-common:[17.0.2].

IT-LiuSir, 2021-01-12:
不知道什么原因造成，我无法解决
`FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-camera:generateGeneralDebugRFile'.
> Could not resolve all files for configuration ':react-native-camera:generalDebugRuntimeClasspath'.
   > Could not resolve com.google.android.gms:play-services-vision-common:[17.0.2].
     Required by:
         project :react-native-camera > com.google.android.gms:play-services-vision:17.0.2
      > Failed to list versions for com.google.android.gms:play-services-vision-common.
         > Unable to load Maven meta-data from https://maven.google.com/com/google/android/gms/play-services-vision-common/maven-metadata.xml.
            > Could not HEAD 'https://maven.google.com/com/google/android/gms/play-services-vision-common/maven-metadata.xml'.
               > Connect to maven.google.com:443 [maven.google.com/216.58.200.46] failed: connect timed out

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org
`

tewarig, 2021-01-12:
check this comment for a while.. does this help out?
https://github.com/react-native-camera/react-native-camera/issues/3064#issuecomment-753489182

IT-LiuSir, 2021-01-13:
我已经解决该问题

closed 2021-01-13 by IT-LiuSir

lunajing, 2021-02-19:
咋解决的 @IT-LiuSir