react-native-camera/react-native-camera #3092: 3.40.0 Memory usage increasing and App get crashed when barcode scanner running

subodha-amarasinghe, 2021-01-21:
# Bug Report

**To Do First**

- [ Yes] Did you try latest release?
- [No ] Did you try master?
- [Yes ] Did you look for existing matching issues?

**Platforms**
iOS

**Versions**

- iOS:14.3
- react-native-camera: 3.40.0
- react-native :0.63.3
- react: 16.13.1
-react-navigation: 4.4.3

**Description/Current Behaviour**

```javascript
renderCamera = () => {
    const { isFlashOn } = this.state;

    return (
      <RNCamera
        ref={this.setCameraRef}
        style={styles.preview}
        type={RNCamera.Constants.Type.back}
        autoFocus={RNCamera.Constants.AutoFocus.on}
        flashMode={
          !this.props.isProductView && (isFlashOn || this.props.torchEnabled)
            ? RNCamera.Constants.FlashMode.torch
            : RNCamera.Constants.FlashMode.off
        }
        androidCameraPermissionOptions={androidPermissionOptions}
        androidRecordAudioPermissionOptions={null}
        googleVisionBarcodeType={
          RNCamera.Constants.GoogleVisionBarcodeDetection.BarcodeType.ALL
        }
        onGoogleVisionBarcodesDetected={this.handleGoogleVisionBarCode}
        googleVisionBarcodeMode={
          RNCamera.Constants.GoogleVisionBarcodeDetection.BarcodeMode.ALTERNATE
        }
        captureAudio={false}
      />
    );
  };

keep barcode scanner camera on.
App memory usage was increasing and then app get crashed after few minutes.

I have attached memory usage graph in Xcode debug navigator

Details
The app “BarCodeReader
![Screenshot 2021-01-21 at 09 29 23](https://user-images.githubusercontent.com/5609203/105278931-82bfdc80-5bcc-11eb-84a0-4c083a091cb8.png)
” on iPhone quit unexpectedly.
Domain: IDEDebugSessionErrorDomain
Code: 4
Failure Reason: Message from debugger: Terminated due to memory issue
--
System Information
macOS Version 11.1 (Build 20C69)
Xcode 12.3 (17715) (Build 12C33)
Timestamp: 2021-01-21T09:30:37+05:30

**Expected Behaviour**

App should not crashed when barcode scanner is on

**Steps to Reproduce**

keep barcode scanner camera on for few minutes.

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

tucao1701, 2021-01-28:
I have same issue when use "onGoogleVisionBarcodesDetected"

aliraza-noon, 2021-01-28:
Facing a similar issue it keeps consuming memory until device run out of memory and the app crashes

jrobichaud, 2021-02-17:
I have a similar problem on iPhone 12 Pro but I noticed if I cover the camera physically the memory finally decrease gradually as the results comes in. 

[image]

It sends a call to firebase every 0.5 seconds and has to keep in memory the image buffer until the results comes in.

After a while in the memory appears to be freed.
[image]

marvintrajano, 2021-04-19:
I'm running into the same issue of running out of memory. My app is also using onGoogleVisionBarcodesDetected. Did anyone find a solution to this?

aliraza-noon, 2021-04-20:
this was an issue for us we tried everything we could eventually we switched to `react-native-camera-kit`

jrobichaud, 2021-04-20:
https://github.com/jrobichaud/react-native-camera/tree/crop-rect-of-interest

I made this fork/branch where I crop the image around the rect of interest before it is sent to firebase which reduce greatly the bandwidth. The original version sends the full image to  firebase every 0.5s. 

To avoid crash a rect of interest of about 20% of the full area should do but you should experiment yourself. The internet speed has an impact whether the image queue will saturate or not.

However you must make sure to align the rect of interest properly with a visual aid. The coordinate system is a bit awkward.

Note: This creates a regression that the coordinates of the scanned barcode are not useable(sorry) but you should not really need it.

closed 2021-06-26 by stale[bot]

waqaramjad, 2022-04-15:
still facing this issue