react-native-camera/react-native-camera #3096: Some Question about rectOfInterest

zjw199011, 2021-01-26:
It's a little bit odd.

I want to limited the scanning area as a 80% width and 100 height rectangle in the center of the view.

const window = Dimensions.get("window");
const rectOfInterest = {
    x: (window.width - window.width * 0.8) / 2 / window.width,
    y: (window.height - 100) / 2 / window.height,
    width: rectWidth / window.width,
    height: rectHeight / window.height,
}
const cameraViewDimensions = { width: window.width, height: window.height}

For IOS: 
I didn't pass the rectOfInterest variable into the RNCamera component, but it only can recongize the barcode in the center of the view.
If I pass the rectOfInterest into the component, it won't recongnize any of the barcode.

For Android:
It works fine without the rectOfInterest variable.

Does anyone use this feature, can you share how to use it?
Or can we have some example of how to using the rectOfInterest and cameraViewDimensions to limit the scanning area???

Is it possible to show the area in the view; then we can easily to test it?

Thanks

eamonn-alphin, 2021-02-12:
This helped me, at least for portrait mode, but I don't understand why we have to set the cameraViewDimensions to the screen width. 

https://stackoverflow.com/questions/49880395/react-native-camera-limit-barcode-scan-area

MarcoScabbiolo, 2021-02-12:
The current behaviour is odd, it will be reviewed and properly documented in v4.
The Android implementation was already reworked and setting `cameraViewDimensions` is no longer necessary.

zjw199011, 2021-02-15:
> The current behaviour is odd, it will be reviewed and properly documented in v4.
> The Android implementation was already reworked and setting `cameraViewDimensions` is no longer necessary.

Sounds good. thanks.

closed 2021-04-23 by stale[bot]