react-native-camera/react-native-camera #3099: Different behaviour on Development vs Release when Microphone disabled

xeroxoid, 2021-02-03:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- iOS: 14.3
- react-native-camera: 3.42.0
- react-native: 0.63.2
- react: 16.13.1
- react-navigation: 4.4.1

**Description/Current Behaviour**

Our code checks for `recordAudioPermissionStatus` when the user is about to record. If the user denies it a prompt appears that disables recording and urges the user to accept (via settings) the permission. i.e. 

```
onStatusChange={({ cameraStatus, recordAudioPermissionStatus }) => {
                this.setState({
                  notAuthorized:
                    cameraStatus === RNCamera.Constants.CameraStatus.NOT_AUTHORIZED ||
                    recordAudioPermissionStatus === RNCamera.Constants.CameraStatus.NOT_AUTHORIZED,
                });
              }}
```

On Development this works as expected and if the user has not enabled permissions, he can not record. On Release though, if the user denies the permission (or changes it from settings in a later stage), `recordAudioPermissionStatus` does not return 'NOT_AUTHORIZED'. So, he records a video with no audio at all.

**Expected Behaviour**

`recordAudioPermissionStatus` should return `NOT_AUTHORIZED` if the permission is denied or disabled via settings. Unfortunately this is not easily testable because it only happens in Release (i.e. Production) schemes.

**Steps to Reproduce**

1. Set scheme to Release
2. Build app
3. Deny permission (or change it in app settings)
4. No prompt
5. Record video with audio
6. No audio recorded :(

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2021-02-08:
I think this is a problem in how Dev vs Prod works.  A long time ago, someone decided it was a good idea to "remove" the code that requests for iOS audio permissions from source code in order to be able to publish the app without audio-recording permission.

Long story short: if Apple detects that your app uses the permission API to request for microphone access, you need to provide the required values in the info.plist file and relevant store locations.  So basically the compiler removes the code at compile time (through macros) in order for people to release apps without microphone permissions if they are not using it.  With this in mind, the permission prompt is fired automatically when the code requests access to the microphone (why apple doesn't detect this as well is beyond me).

This issue has been annoying us for a long time as well, since microphone permission is only requested on recording start when in production, but it is requested on mount in dev. However, "fixing" this seems difficult without breaking everyone's apps that do not use microphone permissions.

MarcoScabbiolo, 2021-02-25:
Will be fixed in v4:

https://github.com/react-native-camera/react-native-camera-v4/blob/05b955627c464e4af0fd4e8fe45726476c851b7b/packages/core/ios/RNCameraManager.swift#L175

closed 2021-05-03 by stale[bot]

xeroxoid, 2021-09-30:
@MarcoScabbiolo has this been merged to the v4 release?