react-native-camera/react-native-camera #3104: Error build when change to mlkit

misaku, 2021-02-09:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

i test only android

**Versions**

- Android: 19
- iOS:
- react-native-camera:3.42;master
- react-native:0.63.4
- react:16.13.1
  

**Description/Current Behaviour**

```
Transform artifact firebase-ml-vision-face-model-17.0.2.aar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) with DexingWithClasspathTransform

> Task :app:checkDebugDuplicateClasses FAILED

AGPBI: {"kind":"error","text":"Type com.google.android.gms.internal.vision.zzhh is referenced as an interface from `com.google.android.gms.internal.vision.zzjx$zza$zza`.","sources":[{}],"tool":"D8"}

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':app:checkDebugDuplicateClasses'.
> 1 exception was raised by workers:
  java.lang.RuntimeException: java.lang.RuntimeException: Duplicate class com.google.android.gms.internal.vision.zzbl found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzbm found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzbn found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzbo found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzbp found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzbr found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zze found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzf found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzg found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzh found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzi found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzjx found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzjy found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzjz found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzka found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzkb found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzkc found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzkd found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzke found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzkf found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzkg found in modules jetified-firebase-ml-vision-face-model-17.0.2-runtime.jar (com.google.firebase:firebase-ml-vision-face-model:17.0.2) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzv found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzw found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  
  Go to the documentation to learn how to <a href="d.android.com/r/tools/classpath-sync-errors">Fix dependency resolution errors</a>.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

* Get more help at https://help.gradle.org

BUILD FAILED in 1m 20s

```
**Additionals**
[... 278 lines cut ...]
  }

  implementation(project(':react-native-device-info')) {
    exclude group: 'com.google.android.gms'
  }

  if (enableHermes) {
    def hermesPath = "../../node_modules/hermes-engine/android/";
    debugImplementation files(hermesPath + "hermes-debug.aar")
    releaseImplementation files(hermesPath + "hermes-release.aar")
  } else {
    implementation jscFlavor
  }
}

// Run this once to be able to run the application with BUCK
// puts all compile dependencies into folder libs for BUCK to use
task copyDownloadableDepsToLibs(type: Copy) {
  from configurations.compile
  into 'libs'
}

apply from: file("../../node_modules/@react-native-community/cli-platform-android/native_modules.gradle"); applyNativeModulesAppBuildGradle(project)

```

flodev, 2021-03-01:
Hi, [this](https://github.com/react-native-camera/react-native-camera/issues/3068#issuecomment-761332714) helped me. I've pasted it and haven't hat the duplicate class error anymore

renatobenks, 2021-05-05:
Did it work @misaku?

closed 2021-05-05 by renatobenks