react-native-camera/react-native-camera #3111: recordAsync() stops recording sound when video running on same screen

SamiChab, 2021-02-12:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [ ] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- iOS: 14.4
- react-native-camera: 3.42.2
- react-native: 0.61.2
- react: 16.9.0

**Description/Current Behaviour**

I'm running a RNCamera view on the same screen as a react-native-video view.
I'm calling recordAsync() to record a video. Everything works fine (no error from recordAsync, neither onAudioInterrupted called), but as soon as I'm playing the react-native-video, the sound stops being captured by the RNCamera module. Same behaviour when I connect headphones.

**Expected Behaviour**

The sound should be captured whether a video is playing or not.

**Steps to Reproduce**

```
<RNCamera
	ref={ref => this.cameraRef = ref}
	style={styles.cameraContainer}
	onBarCodeRead={this.handleBarCodeRead}
	captureAudio={true}
	onCameraReady={() => this.setState({launchingCamera: false})}
	type={RNCamera.Constants.Type.front}
	flashMode={RNCamera.Constants.FlashMode.off}
	onRecordingStart={() => {
		console.log('onRecordingStart')
		setTimeout(() => this.setState({currentState: STATES.playing}), 2000)
		// this plays the react-native-video 2 seconds after the RNCamera recording starts.
		// the final video I get from recordAsync() has 2 seconds of sound at the beginning, then the video continues but without sound
	}}
	onRecordingEnd={() => console.log('onRecordingEnd')}
	onAudioConnected={() => console.log('onAudioConnected')}
	onAudioInterrupted={(data) => console.log('onAudioInterrupted')}
	playSoundOnCapture={false}
	playSoundOnRecord={false}
/>
```

```
this.cameraRef.recordAsync({
	quality: RNCamera.Constants.VideoQuality['480p'],
	orientation: 'portrait',
	mirrorVideo: true,
	mute: false,
	codec: Platform.OS == 'ios' ? RNCamera.Constants.VideoCodec.H264 : undefined,
})
	.then(data => {
		console.log('recordAsync', {data})
		this.props.uploadVideo(data)
	})
	.catch(error => {
		console.log('recordAsync', {error})
		this.setState({cameraRecordError: error})
	})
```

Thanks in advance to anyone that can help 👍

IacoCesar, 2021-02-15:
Same here, 

check this post [https://github.com/react-native-camera/react-native-camera/issues/2803](url)

IacoCesar, 2021-02-15:
@SamiChab

The problem is caused by ignoreSilentSwitch of react-native-video, so i just tried changing the ignore param to a invalid param "ignores" and works fine

SamiChab, 2021-02-16:
@IacoCesar thank you so much !!! It works now :)

closed 2021-02-16 by SamiChab