react-native-camera/react-native-camera #3118: Could not determine the dependencies of task ':app:mergeDebugAssets'.

biswajitmaity2019, 2021-02-21:
# Question

                                                                                                                        Deprecated Gradle features were used in this build, making it incompatible with Gradle 7.0.                             Use '--warning-mode all' to show the individual deprecation warnings.                                                   See https://docs.gradle.org/6.2/userguide/command_line_interface.html#sec:command_line_warnings                                                                                                                                                 FAILURE: Build failed with an exception.                                                                                                                                                                                                        * What went wrong:                                                                                                      Could not determine the dependencies of task ':app:mergeDebugAssets'.                                                   > Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.                                  > Could not find :react-native-camera:.                                                                                   Required by:                                                                                                                project :app                                                                                                                                                                                                                           * Try:                                                                                                                  Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.                                                                                                                                                                                                                    * Get more help at https://help.gradle.org                                                                                                                                                                                                      BUILD FAILED in 8s                                                                                                                                                                                                                              error Failed to install the app. Make sure you have the Android development environment set up: https://reactnative.dev/docs/environment-setup. Run CLI with --verbose flag for more details.                                                   Error: Command failed: gradlew.bat app:installDebug -PreactNativeDevServerPort=8081  

1. 

- 

I am facing this issue, can you solve this?

reyraa, 2021-02-22:
This is the full error description:

Could not determine the dependencies of task ':app:mergeDebugAssets'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve project :react-native-camera.
     Required by:
         project :app
      > Cannot choose between the following variants of project :react-native-camera:
          - generalDebugRuntimeElements
          - mlkitDebugRuntimeElements
        All of them match the consumer attributes:
          - Variant 'generalDebugRuntimeElements' capability Lisk:react-native-camera:unspecified:
              - Unmatched attributes:
                  - Found com.android.build.api.attributes.VariantAttr 'generalDebug' but wasn't required.
                  - Found react-native-camera 'general' but wasn't required.
              - Compatible attributes:
                  - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
                  - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
                  - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.
          - Variant 'mlkitDebugRuntimeElements' capability Lisk:react-native-camera:unspecified:
              - Unmatched attributes:
                  - Found com.android.build.api.attributes.VariantAttr 'mlkitDebug' but wasn't required.
                  - Found react-native-camera 'mlkit' but wasn't required.
              - Compatible attributes:
                  - Required com.android.build.api.attributes.BuildTypeAttr 'debug' and found compatible value 'debug'.
                  - Required com.android.build.gradle.internal.dependency.AndroidTypeAttr 'Aar' and found compatible value 'Aar'.
                  - Required org.gradle.usage 'java-runtime' and found compatible value 'java-runtime'.

Running with
- React Native: 0.63.4
- react-native-camera: 3.42.2
- OS: Mac OS

codeLearnerrr, 2021-02-24:
Solution https://github.com/mikepenz/Android-Iconics/issues/452#issuecomment-637886264

closed 2021-05-02 by stale[bot]

DibyajyotiMishra, 2021-05-25:
In the android/app/build.gradle add the following lines:

android {
     defaultConfig {
                    ..........
                    missingDimensionStrategy 'react-native-camera', 'general'   # This line
  }
}

Piyush132000, 2021-06-12:
```
  compileOptions {
        sourceCompatibility JavaVersion.VERSION_1_8
        targetCompatibility JavaVersion.VERSION_1_8
    }

    defaultConfig {
        applicationId "com.train"
        minSdkVersion rootProject.ext.minSdkVersion
        targetSdkVersion rootProject.ext.targetSdkVersion
        versionCode 1
        versionName "1.0"
        multiDexEnabled true   // add this line 
        missingDimensionStrategy 'react-native-camera', 'general' // add this line 
    }
```

**this work for me **

rahadkc, 2021-07-06:
> In the android/app/build.gradle add the following lines:
> 
> android {
> defaultConfig {
> ..........
> missingDimensionStrategy 'react-native-camera', 'general' # This line
> }
> }

@DibyajyotiMishra It works. Thank you

FoolCoder, 2021-09-16:
I AM GETTING THIS error it was working fine but all of sudden it giving me this i don't know why i did not even use camera in my application

mobilewd, 2021-09-16:
yes me too, getting this error by today
[image]

I think this happens because of the server timeout?

FoolCoder, 2021-09-16:
my server is working properly

DibyajyotiMishra, 2021-09-18:
> I AM GETTING THIS error it was working fine but all of sudden it giving me this i don't know why i did not even use camera in my application
> [image]

Try to clean your gradle and build again.

suraj97489, 2022-01-16:
i am also getting same error

dharmendra841434, 2022-10-31:
<h1> I am Getting this errors anyone can help me </h1>

Could not determine the dependencies of task ':app:mergeDebugAssets'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.   
   > Could not resolve com.facebook.react:react-native:+.
     Required by:
         project :app
         project :app > project :react-native-vector-icons
         project :app > project :react-native-community_datetimepicker
         project :app > project :react-native-firebase_app
         project :app > project :react-native-firebase_auth
         project :app > project :react-native-firebase_database
         project :app > project :react-native-firebase_firestore
         project :app > project :react-native-firebase_storage
         project :app > project :react-native-masked-view_masked-view
         project :app > project :agora-react-native-rtm
         project :app > project :react-native-agora
         project :app > project :react-native-document-picker
         project :app > project :react-native-gesture-handler
         project :app > project :react-native-image-crop-picker
         project :app > project :react-native-pg-react-native-sdk
         project :app > project :react-native-safe-area-context
         project :app > project :react-native-screens
         project :app > project :react-native-webview

anjayluh, 2022-10-31:
> # I am Getting this errors anyone can help me
> Could not determine the dependencies of task ':app:mergeDebugAssets'.
> 
> > Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
> > Could not resolve com.facebook.react:react-native:+.
> > Required by:
> > project :app
> > project :app > project :react-native-vector-icons
> > project :app > project :react-native-community_datetimepicker
> > project :app > project :react-native-firebase_app
> > project :app > project :react-native-firebase_auth
> > project :app > project :react-native-firebase_database
> > project :app > project :react-native-firebase_firestore
> > project :app > project :react-native-firebase_storage
> > project :app > project :react-native-masked-view_masked-view
> > project :app > project :agora-react-native-rtm
> > project :app > project :react-native-agora
> > project :app > project :react-native-document-picker
> > project :app > project :react-native-gesture-handler
> > project :app > project :react-native-image-crop-picker
> > project :app > project :react-native-pg-react-native-sdk
> > project :app > project :react-native-safe-area-context
> > project :app > project :react-native-screens
> > project :app > project :react-native-webview

same here, just started an hour ago too. 😞

Davuttrg, 2022-10-31:
I am also getting the same error. I didn't change anything.

FAILURE: Build failed with an exception.

* What went wrong:
Could not determine the dependencies of task ':app:mergeDebugAssets'.
> Could not resolve all task dependencies for configuration ':app:debugRuntimeClasspath'.
   > Could not resolve com.facebook.react:react-native:+.
     Required by:
         project :app
         project :app > project :tipsi-stripe
         project :app > project :react-native-community_art
         project :app > project :react-native-community_async-storage
         project :app > project :react-native-community_cameraroll
         project :app > project :react-native-community_datetimepicker
         project :app > project :react-native-community_masked-view
         project :app > project :react-native-community_slider
         project :app > project :react-native-fs
         project :app > project :react-native-gesture-handler
         project :app > project :react-native-image-crop-picker
         project :app > project :react-native-image-picker
         project :app > project :react-native-linear-gradient
         project :app > project :react-native-localization
         project :app > project :react-native-reanimated
         project :app > project :react-native-safe-area-context
         project :app > project :react-native-screens
         project :app > project :react-native-share
         project :app > project :react-native-snackbar
         project :app > project :react-native-vector-icons
         project :app > project :react-native-view-shot
         project :app > project :react-native-webview
      > Failed to list versions for com.facebook.react:react-native.
         > Unable to load Maven meta-data from https://jcenter.bintray.com/com/facebook/react/react-native/maven-metadata.xml.
            > Could not HEAD 'https://jcenter.bintray.com/com/facebook/react/react-native/maven-metadata.xml'.
               > Read timed out

Kajanan02, 2022-10-31:
I am also getting the same error. I didn't change anything.

vaibhavsonawane, 2022-10-31:
I am getting the same error since morning, didn't change anything

selvaprasath2706, 2022-10-31:
For Linux / MacOS- ./gradlew clean

Windows PowerShell- .\gradlew clean

Windows cmd- gradlew clean

This worked for me from https://stackoverflow.com/a/69477108

anjayluh, 2022-10-31:
> For Linux / MacOS- ./gradlew clean
> 
> Windows PowerShell- .\gradlew clean
> 
> Windows cmd- gradlew clean
> 
> This worked for me from https://stackoverflow.com/a/69477108

This doesn't fix it for me. I see a jcenter downtime report[ being discussed](https://stackoverflow.com/questions/74258160/is-jcenter-down-permanently-31-oct) that's happening today

OmarUsman777, 2022-10-31:
The jcenter was down for more than 12 hours, Which caused a panic and huge disturbance among android developers including RN. Jcenter is up now, Revert all your changes made to fix this issue and run the code as usual. Happy Coding :)

vaibhavsonawane, 2022-11-01:
> The jcenter was down for more than 12 hours, Which caused a panic and huge disturbance among android developers including RN. Jcenter is up now, Revert all your changes made to fix this issue and run the code as usual. Happy Coding :)

Thanks @OmarUsman777 for your help