react-native-camera/react-native-camera #3120: iOS recordAsync() is not resolving after recording a long video (> 1mn)

SamiChab, 2021-02-22:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- iOS: v14.4 and other versions
- react-native-camera: 3.42.2
- react-native: 0.61.2
- react: 16.9.0

**Description/Current Behaviour**

On iOS, I call `recordAsync()` and wait approx. 1mn. Then,  I call `stopRecording()`. `onRecordingEnd` gets fired but `recordAsync()` doesn't get resolved...

**Expected Behaviour**

`recordAsync()` should resolve after calling `stopRecording()`

**Steps to Reproduce**

On iOS, call `recordAsync()` and wait approx. 1mn. Then, call `stopRecording()`. `recordAsync()` should resolve but it doesn't

Thanks in advance to anyone that can help 👍

longman0512, 2021-02-24:
Hello.
I think you are going to record the screen using camera.
Can you show me the issue in detail?

longman0512, 2021-02-24:
I need the screenshot if you possible.

SamiChab, 2021-02-25:
Hello,
Which details and what screenshot do you need @longman0512 ?

cristianoccazinsp, 2021-02-26:
Which iphone did you test on? Can you also provide the config options you used for both RNCamera and recordAsync?

Can you try with different qualities?  My guess is that the file is too big, and the promise is not resolving, or taking too long to resolve, because the file processing is slow or causes an unexpected error.

SamiChab, 2021-02-27:
@cristianoccazinsp thanks for answering! The issue appears on different devices: iPhone 7; iPhone 8 and 8 Plus. Always iOS 14.4

Here is my code:

```
<RNCamera
	ref={ref => this.cameraRef = ref}
	style={styles.cameraContainer}
	captureAudio={true}
	onCameraReady={() => this.setState({launchingCamera: false})}
	type={RNCamera.Constants.Type.front}
	flashMode={RNCamera.Constants.FlashMode.off}
	onRecordingStart={() => /* ... */ }
	onRecordingEnd={() => /* ... */ }
	onAudioConnected={() => /* ... */ }
	onAudioInterrupted={() => /* ... */ }
	playSoundOnCapture={false}
	playSoundOnRecord={false}
/>

```

```
this.cameraRef.recordAsync({
	quality: RNCamera.Constants.VideoQuality['480p'],
	orientation: 'portrait',
	mirrorVideo: Platform.OS == 'ios', // allows me to get a .mp4 video on iOS (it's .mov otherwise)
	mute: false,
	codec: Platform.OS == 'ios' ? RNCamera.Constants.VideoCodec.H264 : undefined,
})
	.then(data => { /* ... */ })
	.catch(error => { /* ... */ })
```

cristianoccazinsp, 2021-02-27:
Try removing the mirror and orientation flags first to see if the issue
still happens. I would bet the mirroring is causing issues.

SamiChab, 2021-02-27:
@cristianoccazinsp thank you so much! Deleting `mirrorVideo` solved the problem!!

I leave this issue open, because `recordAsync()` should work on long videos with `mirrorVideo` enabled.

closed 2021-05-06 by stale[bot]