react-native-camera/react-native-camera #3122: Real-time Automatic Deep Matting For Mobile Devices

simvirakaiml, 2021-02-23:
# Feature-Request

**Describe the Feature**

As a user, I want to change camera background with a selected image during living camera session and recording it.
Example: https://3.bp.blogspot.com/-jp16CE_SLZk/WpW1sKWU9PI/AAAAAAAACY0/sjHghHiuarEC2aEy5txhmdT6INK9C_OxACLcBGAs/s640/image3.gif

**Possible Implementations**

Android: See [Portrait Segmentation](https://github.com/anilsathyan7/Portrait-Segmentation) for more information.

iOS: _Not yet found!_

**Related Issues**

> Love react-native-camera? Please consider supporting our collective: 👉  https://opencollective.com/react-native-camera/donate
> Want this feature to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

MarcoScabbiolo, 2021-02-23:
This is quite a massive feature and a very specific one. The only way I see it being viably implemented is as plugin in v4. To be honest there is a ton of work to be made before we can even consider contributing ourselves for this kind of feature, not to mention the work needed to get a stable extensible version that supports this kind of feature as an external addition.

Do you know of a provider or trained ml model small enough to fit into a mobile device that can detect the head. I has to be in-device to be able to work in real time.

Now there is one side of the feature that looks like a good candidate for a popular plugin: live streaming the video. An HLS/RTMP upload plugin can support most use cases and backend providers.

simvirakaiml, 2021-02-23:
Hello, @MarcoScabbiolo. Sure, I bet this is a massive one, especially diving into native code it's a big challenge for me. Anyway it toke almost several weeks for me to integrate with react-native-camera. Switching between normal camera and replace background with fixed image is working but not smooth enough I think. And recording is not works yet. If you would like to see, I am very happy to prepare a new repo for you.

closed 2021-05-01 by stale[bot]