react-native-camera/react-native-camera #3124: Inconsistent barcode scanning on iOS. Works fine on Android.

eteamIH, 2021-02-23:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**
iOS

**Versions**

- Android: 10 (Samsung Galaxy A20)
- iOS: 12 (iPhone 6)
- react-native-camera: 3.42.
- react-native: 0.63

**Description/Current Behaviour**

Many types of barcodes are scanned on Android, including those not listed in the docs (RSS_EXPANDED) but on iOS it only scans a limited number of barcodes.

**Expected Behaviour**

Barcodes that get scanned normally on Android should also get scanned without any issue on iOS.

**Steps to Reproduce**

Scan a GS1 expanded coupon barcode. Works perfectly on Android, not working at all on iOS.

Scan an EAN-13 barcode. Works perfectly on Android. iOS sometimes doesn't scan at all, sometimes takes a lot of time to scan, sometimes scans when the barcode is placed vertically.

Scan an EAN-8 barcode. Works perfectly on Android. iOS sometimes doesn't scan at all, sometimes takes a lot of time to scan, sometimes scans when the barcode is placed vertically.

eteamIH, 2021-03-15:
Turns out GS1 Databar Expanded, a.k.a RSS Expanded barcodes are not supported to be scanned on iOS. 

We tried using another library [Zxingify-ObjectiveC](https://github.com/zxingify/zxingify-objc) as a native module but its performance is not satisfactory on newer iPhones (11 Pro, 12), scanning success rate is around 15%. Works fine on older ones (iPhone 6, 6s).

We are currently exploring [react-native-barcode](https://github.com/yvvw/react-native-barcode) which works fine when scanning from an image, but the camera scanner doesn't support RSS Expanded barcodes.

If anyone ends up in a similar problem and comes up with a better solution, please mention the details here.

Thanks

perfectcircle, 2021-03-17:
We do experience the same problems. We tested scanning EAN-13 barcodes with 6 different iPhones (older models seem to work better than newer models) and it seems like the do struggle a lot whenever the lines are printed out a bit blurry - or if printed out large (15 cm width). And it is very sensitive to lighting conditions. Android worked like a charm.

closed 2021-05-23 by stale[bot]

yorickshan, 2021-12-30:
same issue

MarufSharifi, 2022-07-30:
same issue