react-native-camera/react-native-camera #3126: Type for "cameraId" is undefined

lxender, 2021-02-24:
# Bug Report

The property "cameraId" is undefined in index.d.ts but is accepted by the component.

[image]

cristianoccazinsp, 2021-02-26:
Looks like a typings/ts missing prop? Can you provide a PR to include it in `index.d.tx` ? :)

closed 2021-05-05 by stale[bot]

n1ru4l, 2021-05-12:
:tada: This issue has been resolved in version 3.43.8 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera/v/3.43.8)
- [GitHub release](https://github.com/react-native-camera/react-native-camera/releases/tag/v3.43.8)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket:

PR #3196 (2021-05-12): fix(type definitions for typescript): add type to missing prop `cameraId` on camera component
fix #3126