react-native-camera/react-native-camera #3132: No visible @interface for 'FIRVisionFaceDetector' declares the selector 'processImage:completion:'

ShivaSunny9, 2021-02-26:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

iOS

**Versions**

- Android:
- iOS: 12.4V
- react-native-camera:^3.31.1
- react-native:^0.58.0
- react:16.6.3
  

**Description/Current Behaviour**

my pod file 

# Uncomment the next line to define a global platform for your project
platform :ios, '10.0'

target 'myearthid' do
  # Comment the next line if you don't want to use dynamic frameworks
# use_frameworks!

  # Pods for myearthid
pod 'web3.swift'
pod 'Firebase/Analytics'
 pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
   'FaceDetectorMLKit'
 ]
pod 'GoogleMLKit/FaceDetection'

  target 'myearthidTests' do
    inherit! :search_paths
    # Pods for testing
  end

end

post_install do |installer|
  installer.pods_project.targets.each do |target|
    target.build_configurations.each do |config|
     config.build_settings['IPHONEOS_DEPLOYMENT_TARGET'] = '10.0'
    
  require 'fileutils'

  FileUtils.rm_rf('Pods/JSONWebToken/CommonCrypto')
    end
   end
end

After running build getting this error 
react-native-camera/ios/RN/FaceDetectorManagerMlkit.m:118:6: No visible @interface for 'FIRVisionFaceDetector' declares the selector 'processImage:completion:'

my FaceDetectorManagerMlKit.m file
#import "FaceDetectorManagerMlkit.h"
#import <React/RCTConvert.h>
#if __has_include(<FirebaseMLVision/FirebaseMLVision.h>)

@interface FaceDetectorManagerMlkit ()
@property(nonatomic, strong) FIRVisionFaceDetector *faceRecognizer;
@property(nonatomic, strong) FIRVision *vision;
[... 311 lines cut ...]
             @"Mode" : @{},
             @"Landmarks" : @{},
             @"Classifications" : @{}
             };
}

@end
#endif

**Expected Behaviour**

**Steps to Reproduce**

Integrated firbase and installed  
pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
   'FaceDetectorMLKit'
 ]

when running the build getting this error
[image]

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

renatobenks, 2021-04-28:
is this still happening @ShivaSunny9 or can we close it? Thank you for reporting!

ShivaSunny9, 2021-04-29:
@renatobenks it is still happening, not yet resolved.

closed 2021-07-05 by stale[bot]