react-native-camera/react-native-camera #3133: Android build issue with new release (3.43.0)

techericb, 2021-02-26:
# Bug Report

**To Do First**

- [x ] Did you try latest release?
- [ ] Did you try master?
- [x ] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- react-native-camera: 3.43.0
- react-native: 0.61.5
- react: 16.14

**Description/Current Behaviour**

Yesterday (2/26) the build worked fine with version 3.42.3
Now the build fails with following error:
/Users/runner/work/1/s/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:536: error: cannot find symbol
            retriever.close();
                     ^
  symbol:   method close()
  location: variable retriever of type MediaMetadataRetriever

**Expected Behaviour**

Build succeeds

**Steps to Reproduce**

Build a project using new version of RN-camera.
Note:  built on App Center.

**Additionals**

cristianoccazinsp, 2021-02-27:
Need to bump compile SDK from 28 to 29. I believe this should have been included in the release notes!

jayfar, 2021-03-04:
Slight sidebar here... this is [perhaps] an example of my frustration or lack of understanding with the npm system. Is this considered a "breaking" change?

We have had this in our package.json file for a long while and our auto build pipelines have been happy until today. 
"react-native-camera": "^3.11.1"

The react-native-camera is a very small part of our app that does not get much attention.

I guess the best practice to prevent this is to use a fixed version in the package json and only test package updates periodically and manually?

To just get my build to succeed, I just updated the package.json to this to avoid dealing with this now.
"react-native-camera": "3.42.3"

closed 2021-05-11 by stale[bot]