react-native-camera/react-native-camera #3134: Task :react-native-camera:compileGeneralDebugJavaWithJavac FAILED

abubakarakram167, 2021-02-28:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**

**Expected Behaviour**

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

> Task :react-native-camera:compileGeneralDebugJavaWithJavac FAILED

Deprecated Gradle features were used in this build, making it incompatible with Gradle 6.0.
Use '--warning-mode all' to show the individual deprecation warnings.
See https://docs.gradle.org/5.4.1/userguide/command_line_interface.html#sec:command_line_warnings
76 actionable tasks: 73 executed, 3 up-to-date
/Users/abubakarakram/development/projects/TestMyDesk/MyDearest_mb_front/node_modules/react-native-camera/android/src/main/java/org/reactnative/camera/CameraModule.java:536: error: cannot find symbol
            retriever.close();
                     ^
  symbol:   method close()
  location: variable retriever of type MediaMetadataRetriever
Note: Some input files use or override a deprecated API.
Note: Recompile with -Xlint:deprecation for details.
Note: /Users/abubakarakram/development/projects/TestMyDesk/MyDearest_mb_front/node_modules/react-native-camera/android/src/main/java/com/google/android/cameraview/Camera2.java uses unchecked or unsafe operations.
Note: Recompile with -Xlint:unchecked for details.
1 error

FAILURE: Build failed with an exception.

* What went wrong:
Execution failed for task ':react-native-camera:compileGeneralDebugJavaWithJavac'.
> Compilation failed; see the compiler error output for details.

* Try:
Run with --stacktrace option to get the stack trace. Run with --info or --debug option to get more log output. Run with --scan to get full insights.

abubakarakram167, 2021-02-28:
Can somebody help me with this issue ?

qualitymanifest, 2021-03-01:
Probably related to [this](https://github.com/react-native-camera/react-native-camera/issues/3133), I was having the same issue and solved it by downgrading to 3.42.3 so I haven't tried that fix yet

cristianoccazinsp, 2021-03-01:
Upgrade Android compile SDK from 28 to 29. This change is also required if you plan on keeping up-to-date with react native and being able to release to the android store, and starting August, SDK 30 will be mandatory.

MateusAndrade, 2021-03-01:
@cristianoccazinsp do you think we should update our docs regarding this? We can add this here: https://react-native-camera.github.io/react-native-camera/docs/qa.

cristianoccazinsp, 2021-03-01:
That's probably a good idea. Also those QA seem outdated; there are references to `com.android.support` (now androidx).

However, SDK 29 is the currently "required" (or soon to be no longer supported) SDK by google (https://developer.android.com/distribute/best-practices/develop/target-sdk) and even 29 will no longer be supported shortly, so everyone should be upgrading if still using 28.

abubakarakram167, 2021-03-01:
> Upgrade Android compile SDK from 28 to 29. This change is also required if you plan on keeping up-to-date with react native and being able to release to the android store, and starting August, SDK 30 will be mandatory.

But even if I change it 28 to 29 compile SDK version it again appears with the same error. I have stuck on this issue for 2 days and I have to deliver my work but due to this issue I am still stuck in a problem even after two days.

cristianoccazinsp, 2021-03-01:
@abubakarakram167 did you clean the build folder? It should be working if you've upgraded from SDK 28 to 29.  Do you have the SDK installed?

HeyShinner, 2021-03-02:
"react-native-camera": "^3.23.1" is ok

kamaal111, 2021-03-02:
@cristianoccazinsp I am already using target SDK version 29, but still no success. As state here I am using "react-native-camera": "3.42.3"

MateusAndrade, 2021-03-02:
> @cristianoccazinsp I am already using target SDK version 29, but still no success. As state here I am using "react-native-camera": "3.42.3"

Can you share the android compile version that you are using?

cristianoccazinsp, 2021-03-02:
@kamaal111 that's strange.  The tests run fine, the example apps work fine, and it works fine for me locally as well.  There has to be a reason it is giving you that error even with SDK 29.

Can you try changing these to match the example app to see what's missing? Except minSdkVersion, since you're probably using 20 or 21 if it's a large project.

```
buildscript {
    ext {
        buildToolsVersion = "29.0.2"
        minSdkVersion = 16
        compileSdkVersion = 29
        targetSdkVersion = 29
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath('com.android.tools.build:gradle:4.1.0')

        // NOTE: Do not place your application dependencies here; they belong
        // in the individual module build.gradle files
    }
}
```

oussamabouchareb, 2021-03-03:
> @kamaal111 that's strange. The tests run fine, the example apps work fine, and it works fine for me locally as well. There has to be a reason it is giving you that error even with SDK 29.
>
> Can you [...]

if I do these changes will this affect the support for pre android 10 device?

cristianoccazinsp, 2021-03-03:
No, as long as you don't change the minSdk.

minSdk: android minimum supported version
target/compile sdk: compile tools used to build the app and available features.

Google increases the minimum compile/target sdk from time to time, and you are not allowed to release android apps that do not meet this restriction.

closed 2021-05-09 by stale[bot]

rodrigofbm, 2021-11-02:
I'm facing this error