react-native-camera/react-native-camera #3135: defaultVideoQuality has no effect on android

qualitymanifest, 2021-03-01:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: API 25
- iOS:
- react-native-camera: 3.42.3 (I tried 3.43.0 but had build issues, downgrading fixed them, idk why)
- react-native: 0.60.6
- react: 16.8.6
  

**Description/Current Behaviour**

Specifying `defaultVideoQuality={RNCamera.Constants.VideoQuality['480p']}` has no effect on android. The files I upload are still 720p when i check on my computer. It works correctly on iOS

**Expected Behaviour**

Videos should be the resolution specified regardless of platform, as far as I can tell the RNC docs indicate that it should be supported on Android?

**Steps to Reproduce**

**Additionals**

 I have also tried specifying `quality: RNCamera.Constants.VideoQuality['480p']` in my options object provided to `recordAsync()`, but no difference

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

qualitymanifest, 2021-03-01:
It might help if there was a `getSupportedVideoSizes` method exposed to return [these values](https://github.com/react-native-camera/react-native-camera/blob/42388be530f4a791ff6822b430b346248c2a40af/android/src/main/java/com/lwansbrough/RCTCamera/RCTCamera.java#L153). It's possible the video size I am specifying just isn't compatible with the devices I'm testing it on. But if I knew what sizes were supported, I could choose the closest one that is supported.

fabriziobertoglio1987, 2021-03-31:
https://github.com/react-native-camera/react-native-camera/blob/c591e9d4e07fdd973e8c1cf681d2c5e10025aa0f/android/src/main/java/com/lwansbrough/RCTCamera/RCTCamera.java#L277-L287

https://github.com/react-native-camera/react-native-camera/blob/c591e9d4e07fdd973e8c1cf681d2c5e10025aa0f/android/src/main/java/com/lwansbrough/RCTCamera/RCTCameraModule.java#L252-L266

closed 2021-06-07 by stale[bot]

swushi, 2021-07-19:
Still an issue on 3.44.3