react-native-camera/react-native-camera #3138: Attempt to invoke virtual method 'java.util.List android.hardware.Camera$Parameters.getSupportedFlashModes()' on a null object reference

LaVielle, 2021-03-02:
# Bug Report
I'm seeing crash logs in Sentry with the following error:
```
Attempt to invoke virtual method 'java.util.List android.hardware.Camera$Parameters.getSupportedFlashModes()' on a null object reference
```
I am not able to reproduce this. From what I understand, the error message suggests the native component is trying to access getSupportedFlashModes before the Camera component exists. So I tried adding a ref to the camera, and doing flash related logic only if the ref exists, but I'm still seeing these error reports :(

Here is how I mount RN Camera:
```JavaScript
const cameraRef = useRef<RNCamera>(null)
const [isFrontCamera, setIsFrontCamera] = useState(false)
const [isTorchOn, setIsTorchOn] = useState(false)

const cameraAttributes = Platform.select({
  ios: {
    onBarCodeRead: onBarCodeRead,
  },
  android: {
    onGoogleVisionBarcodesDetected: onGoogleBarcodeDetected,
  },
})

<Camera
  ref={cameraRef}
  captureAudio={false}
  type={isFrontCamera ? RNCamera.Constants.Type.front : RNCamera.Constants.Type.back}
  // check if cameraRef exists before doing any flash stuff => still crashing
  flashMode={cameraRef.current && isTorchOn ? RNCamera.Constants.FlashMode.torch : RNCamera.Constants.FlashMode.off}
  {...cameraAttributes}
/>
```

After a fair amount of googling I haven't found a solution. My native Android knowledge is pretty limited so I don't know how to fix this. I attached the stack trace bellow.

**To Do First**

- [x] Did you try latest release?
- [ ] Did you try master? (=> have not tried since I cannot reproduce the error in the current release)
- [x] Did you look for existing matching issues?

**Platforms**

This is happening only on Android (at least I don't see any crash reports for iOS)

**Versions**

- Android: 10, 9 and 8 (from crash reports)
- iOS: N/A
- react-native-camera: 3.42.0
- react-native: 0.63.4 (also 0.61.5)
- react: 16.13.1
- @react-navigation/native: 5.9.2
- @react-navigation/stack: 5.14.2

**Description/Current Behaviour**

App crashes.

**Expected Behaviour**

App not crashing 🙃

**Steps to Reproduce**

I was not able to reproduce this :(

**Additional**

Here is the trace:

```
java.lang.NullPointerException: Attempt to invoke virtual method 'java.util.List android.hardware.Camera$Parameters.getSupportedFlashModes()' on a null object reference
    at com.google.android.cameraview.Camera1.setFlashInternal
    at com.google.android.cameraview.Camera1.b
    at com.google.android.cameraview.CameraView.setFlash
    at org.reactnative.camera.CameraViewManager.setFlashMode
    at java.lang.reflect.Method.invoke(Method.java)
    at com.facebook.react.uimanager.ViewManagersPropertyCache$PropSetter.updateViewProp
    at com.facebook.react.uimanager.ViewManagerPropertyUpdater$FallbackViewManagerSetter.setProperty
    at com.facebook.react.uimanager.ViewManagerPropertyUpdater.updateProps
    at com.facebook.react.uimanager.ViewManager.updateProperties
    at com.facebook.react.uimanager.NativeViewHierarchyManager.updateProperties
    at com.facebook.react.uimanager.UIViewOperationQueue$UpdatePropertiesOperation.execute
    at com.facebook.react.uimanager.UIViewOperationQueue$1.run
    at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches
    at com.facebook.react.uimanager.UIViewOperationQueue.k
    at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded
    at com.facebook.react.uimanager.GuardedFrameCallback.doFrame
    at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame
    at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame
    at android.view.Choreographer$CallbackRecord.run(Choreographer.java:965)
    at android.view.Choreographer.doCallbacks(Choreographer.java:791)
    at android.view.Choreographer.doFrame(Choreographer.java:722)
    at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:952)
    at android.os.Handler.handleCallback(Handler.java:883)
    at android.os.Handler.dispatchMessage(Handler.java:100)
    at android.os.Looper.loop(Looper.java:214)
    at android.app.ActivityThread.main(ActivityThread.java:7386)
    at java.lang.reflect.Method.invoke(Method.java)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:980)
java.lang.reflect.InvocationTargetException: null
    at java.lang.reflect.Method.invoke(Method.java)
    at com.facebook.react.uimanager.ViewManagersPropertyCache$PropSetter.updateViewProp
    at com.facebook.react.uimanager.ViewManagerPropertyUpdater$FallbackViewManagerSetter.setProperty
    at com.facebook.react.uimanager.ViewManagerPropertyUpdater.updateProps
    at com.facebook.react.uimanager.ViewManager.updateProperties
    at com.facebook.react.uimanager.NativeViewHierarchyManager.updateProperties
    at com.facebook.react.uimanager.UIViewOperationQueue$UpdatePropertiesOperation.execute
    at com.facebook.react.uimanager.UIViewOperationQueue$1.run
    at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches
    at com.facebook.react.uimanager.UIViewOperationQueue.k
    at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded
    at com.facebook.react.uimanager.GuardedFrameCallback.doFrame
    at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame
    at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame
    at android.view.Choreographer$CallbackRecord.run(Choreographer.java:965)
    at android.view.Choreographer.doCallbacks(Choreographer.java:791)
    at android.view.Choreographer.doFrame(Choreographer.java:722)
    at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:952)
    at android.os.Handler.handleCallback(Handler.java:883)
    at android.os.Handler.dispatchMessage(Handler.java:100)
    at android.os.Looper.loop(Looper.java:214)
    at android.app.ActivityThread.main(ActivityThread.java:7386)
    at java.lang.reflect.Method.invoke(Method.java)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:980)
com.facebook.react.bridge.JSApplicationIllegalArgumentException: Error while updating property 'flashMode' of a view managed by: RNCamera
    at com.facebook.react.uimanager.ViewManagersPropertyCache$PropSetter.updateViewProp
    at com.facebook.react.uimanager.ViewManagerPropertyUpdater$FallbackViewManagerSetter.setProperty
    at com.facebook.react.uimanager.ViewManagerPropertyUpdater.updateProps
    at com.facebook.react.uimanager.ViewManager.updateProperties
    at com.facebook.react.uimanager.NativeViewHierarchyManager.updateProperties
    at com.facebook.react.uimanager.UIViewOperationQueue$UpdatePropertiesOperation.execute
    at com.facebook.react.uimanager.UIViewOperationQueue$1.run
    at com.facebook.react.uimanager.UIViewOperationQueue.flushPendingBatches
    at com.facebook.react.uimanager.UIViewOperationQueue.k
    at com.facebook.react.uimanager.UIViewOperationQueue$DispatchUIFrameCallback.doFrameGuarded
    at com.facebook.react.uimanager.GuardedFrameCallback.doFrame
    at com.facebook.react.modules.core.ReactChoreographer$ReactChoreographerDispatcher.doFrame
    at com.facebook.react.modules.core.ChoreographerCompat$FrameCallback$1.doFrame
    at android.view.Choreographer$CallbackRecord.run(Choreographer.java:965)
    at android.view.Choreographer.doCallbacks(Choreographer.java:791)
    at android.view.Choreographer.doFrame(Choreographer.java:722)
    at android.view.Choreographer$FrameDisplayEventReceiver.run(Choreographer.java:952)
    at android.os.Handler.handleCallback(Handler.java:883)
    at android.os.Handler.dispatchMessage(Handler.java:100)
    at android.os.Looper.loop(Looper.java:214)
    at android.app.ActivityThread.main(ActivityThread.java:7386)
    at java.lang.reflect.Method.invoke(Method.java)
    at com.android.internal.os.RuntimeInit$MethodAndArgsCaller.run(RuntimeInit.java:492)
    at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:980)
```

cristianoccazinsp, 2021-03-04:
Crash log points to this line: https://github.com/react-native-camera/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L1443 with `mCameraParameters` being null.

Now that's strange, since `isCameraOpened()` is first checked, which only returns true if the camera is not null (https://github.com/react-native-camera/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L377) and if the camera is not null, the `mCameraParameters` should also not be null because of this (https://github.com/react-native-camera/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L1051)

I can only guess that there's a race condition there in `setFlash`, but if that were the case, pretty much all props should be throwing this exception. I haven't seen it myself at all.

If you want to give it a try, edit this line here in Java https://github.com/react-native-camera/react-native-camera/blob/master/android/src/main/java/com/google/android/cameraview/Camera1.java#L378 and add an extra `&& mCameraParameters != null`

LaVielle, 2021-03-16:
Thanks for looking into this. I forked react-native-camera and made you suggested changes. Works fine in dev. I will put this in production and see if the crash logs disappear.

LaVielle, 2021-05-01:
As a quick update in case anyone is looking at this issue:
the fix suggested by @cristianoccazinsp works. I have had this in production for over a month and the error logs are gone.

I will make a PR when I have time.

closed 2021-07-07 by stale[bot]