react-native-camera/react-native-camera #3143: Failed: "Camera handle cannot be null"

colantuomo, 2021-03-10:
# Question

Hey,
I am trying to test RNCamera as follows:
```Typescript
  describe('when take the picture', () => {
    it('should dispatch onPictureTaken', async () => {
      const onPictureTaken = jest.fn();
      const { getByTestId } = mountWithAtworkHost(
        <Camera
          description="test description"
          onPictureTaken={onPictureTaken}
        />,
      );
      const button = getByTestId('shoot-btn-test-id');
      await act(() => fireEvent.press(button));
      expect(onPictureTaken).toHaveBeenCalledTimes(1);
    });
  });
```

In my component there's a method:
```Typescript
  const takePicture = async () => {
    if (cam) {
      const data = await cam.takePictureAsync(options);
      onPictureTaken(data);
    }
  };
```
It works normally, however in the test I got the error: `Failed: "Camera handle cannot be null"` 

In my component I am already setting his "ref":
```Typescript
const [cam, setCamera] = useState<RNCamera | null>();
 <RNCamera
          style={{ flex: 1 }}
          type={cameraType}
          ref={setCamera}
          captureAudio={false}
        />
```

Does anyone know how I can solve this?

Thanks!

fabriziobertoglio1987, 2021-03-31:
https://github.com/react-native-camera/react-native-camera/blob/ed84c27ff840f3d499377645b4e314c43cd39059/src/RNCamera.js#L732

https://github.com/facebook/react-native/commit/613ca146122bf235aea110b436dbe17ba0912a1d

```
this._cameraHandle = ReactNative.findNodeHandle(ref);
```

>This just greps for React.findNodeHandle and replace them with ReactNative.findNodeHandle. I fixed up the imports manually.
>I also found two callers each of ReactNative.createClass and React.render with the exception of downstream and examples will fix them separately.
>I'll need to find more things like `var { PropTypes } = ReactNative;` separately. I think this is a good start though.

I'll prepare a pr thanks

fabriziobertoglio1987, 2021-03-31:
https://github.com/react-native-camera/react-native-camera/blob/ed84c27ff840f3d499377645b4e314c43cd39059/src/RNCamera.js#L729-L737

Posted by The Saviour https://github.com/necolas/react-native-web/issues/1957#issuecomment-806010607
> believe you can call scrollviewref.getNativeScrollRef().measureLayout(...) to use the non-deprecated API. >Using findNodeHandle here is deprecated.
>https://github.com/facebook/react-native/blob/master/Libraries/Components/ScrollView/ScrollView.js#L892

colantuomo, 2021-03-31:
Hey @fabriziobertoglio1987 thanks for the reply.
Sorry but i didnt understand very much

This means will be a fix in a next version of RN Camera?

fabriziobertoglio1987, 2021-03-31:
@colantuomo feel free to send a pr otherwise I will try to do it in my free time :peace_symbol: :pray:

closed 2021-06-07 by stale[bot]