react-native-camera/react-native-camera #3144: onGoogleVisionBardcodesDetected does not work on Android emulator

colin-peter, 2021-03-11:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:
- react-native:
- react:
  

**Description/Current Behaviour**

The `onGoogleVisionBarcodesDetected` props doesn't trigger the callback in Android Emulator. However, `onBarcodeRead` does. The code works on device so not sure what's going on.
**Expected Behaviour**
`onGoogleVisionBarcodesDetected` should trigger the callback in emulator

**Steps to Reproduce**

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

martmists-gh, 2021-03-31:
Same issue here, though it should be noted that onBarcodeRead does not work for QR codes and as such I am unable to test my code.

fabriziobertoglio1987, 2021-03-31:
https://github.com/react-native-camera/react-native-camera/blob/c591e9d4e07fdd973e8c1cf681d2c5e10025aa0f/android/src/main/java/org/reactnative/camera/RNCameraView.java#L203-L206

closed 2021-06-07 by stale[bot]