react-native-camera/react-native-camera #3145: Android Build issue(version mismatch with latest firebase sdk) when using mlKit

Shahulul, 2021-03-11:
# Bug Report

**To Do First**

- [ ] Did you try latest release?
- [ ] Did you try master?
- [ ] Did you look for existing matching issues?

**Platforms**

**Versions**

- Android:
- iOS:
- react-native-camera:3.43.0
- react-native:0.62.2
- react:16.11.0
- @react-native-firebase/app: "8.4.7",
  
Android build getting failed with the following version mismatch error

* What went wrong:
Execution failed for task ':app:checkDevelopmentDebugDuplicateClasses'.
> 1 exception was raised by workers:
  java.lang.RuntimeException: Duplicate class com.google.android.gms.internal.vision.zzbl found in modules play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0) and play-services-vision-face-contour-internal-16.0.0-runtime.jar (com.google.android.gms:play-services-vision-face-contour-internal:16.0.0)
  Duplicate class com.google.android.gms.internal.vision.zzbn found in modules play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0) and play-services-vision-face-contour-internal-16.0.0-runtime.jar (com.google.android.gms:play-services-vision-face-contour-internal:16.0.0)
  Duplicate class com.google.android.gms.internal.vision.zzbo found in modules play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0) and play-services-vision-face-contour-internal-16.0.0-runtime.jar (com.google.android.gms:play-services-vision-face-contour-internal:16.0.0)
  Duplicate class com.google.android.gms.internal.vision.zzbp found in modules play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0) and play-services-vision-face-contour-internal-16.0.0-runtime.jar (com.google.android.gms:play-services-vision-face-contour-internal:16.0.0)
  Duplicate class com.google.android.gms.internal.vision.zzbr found in modules play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0) and play-services-vision-face-contour-internal-16.0.0-runtime.jar (com.google.android.gms:play-services-vision-face-contour-internal:16.0.0)
  Duplicate class com.google.android.gms.internal.vision.zze found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzf found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzg found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzh found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzi found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzkf found in modules play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0) and play-services-vision-face-contour-internal-16.0.0-runtime.jar (com.google.android.gms:play-services-vision-face-contour-internal:16.0.0)
  Duplicate class com.google.android.gms.internal.vision.zzkh found in modules play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0) and play-services-vision-face-contour-internal-16.0.0-runtime.jar (com.google.android.gms:play-services-vision-face-contour-internal:16.0.0)
  Duplicate class com.google.android.gms.internal.vision.zzki found in modules play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0) and play-services-vision-face-contour-internal-16.0.0-runtime.jar (com.google.android.gms:play-services-vision-face-contour-internal:16.0.0)
  Duplicate class com.google.android.gms.internal.vision.zzkj found in modules play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0) and play-services-vision-face-contour-internal-16.0.0-runtime.jar (com.google.android.gms:play-services-vision-face-contour-internal:16.0.0)
  Duplicate class com.google.android.gms.internal.vision.zzkl found in modules play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0) and play-services-vision-face-contour-internal-16.0.0-runtime.jar (com.google.android.gms:play-services-vision-face-contour-internal:16.0.0)
  Duplicate class com.google.android.gms.internal.vision.zzv found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)
  Duplicate class com.google.android.gms.internal.vision.zzw found in modules play-services-vision-20.0.0-runtime.jar (com.google.android.gms:play-services-vision:20.0.0) and play-services-vision-common-19.1.0-runtime.jar (com.google.android.gms:play-services-vision-common:19.1.0)

It should work without build failure by the latest version of firebase sdk in android

**Steps to Reproduce**

Just rebuild the android project

**Additionals**

[image]

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

fabriziobertoglio1987, 2021-03-31:
https://github.com/react-native-camera/react-native-camera/issues/3031#issuecomment-732208894
https://github.com/react-native-camera/react-native-camera/issues/3104#issuecomment-787803876

tehong, 2021-04-15:
Try this in your app gradle:

``` 
dependencies {

  .
  .
  .
  
  implementation (project(':react-native-camera')) {
          exclude group: "com.google.android.gms"
          exclude group: "com.google.android.gms", module: "play-services-vision"
          exclude group: "com.google.firebase", module: "firebase-ml-vision-face-model"
  }
  implementation ('com.google.android.gms:play-services-vision:20.1.3') {
          force = true
  }
  implementation 'com.google.firebase:firebase-ml-vision:'
  .
  .
  .

}
```

Shahulul, 2021-04-16:
Worked, But there is an incompatibility with the latest firebase SDK. Gives an error when doing npm install and pod install.

closed 2021-05-05 by renatobenks