react-native-camera/react-native-camera #3151: Camera and photos are slightly zoomed

MrHjalmarsson, 2021-03-19:
# Bug Report

**Platforms**

iOS

**Versions**
- iOS: 
- react-native-camera: 3.42.0, 3.43.0
- react-native: 0.63.3
- react: 16.13.1

**Description/Current Behaviour**

Compared to native iOS camera both the camera canvas and the result photos after capturing are slightly zoomed.

**Expected Behaviour**

Photos and canvas to not be zoomed.

**Steps to Reproduce**

Take photo with react-native-camera.
I'm using ratio 1:1.

**Additionals**
Attaching two images, one of photo with native iOS camera and one with react-native-camera

[image]
[image]

cristianoccazinsp, 2021-03-22:
How are you setting the aspect ratio to 1:1? Remember that iOS only captures in 4:3, but the final picture is based on the View's dimensions you've set. So if you render the component in a view that is not exactly 1:1, the photo will end up in another aspect ratio.

MrHjalmarsson, 2021-03-23:
@cristianoccazinsp I set the camera view dimensions to screen width and I set ratio to 1:1 like this:
`style={{ width: screenWidth, height: screenHeight }}`
`ratio={ '1:1' }`

Downgrading to 3.40.0 solves the issue for me.

cristianoccazinsp, 2021-03-23:
I still don't know if the ratio prop works in iOS, I think it's only for
Android photos!

But your problem is right there, you are using screen width and screen
height. For 1:1, you should use screen width for both width and height
style.

In curious about what changed within those versions, but for me, using
styles has always been the only way to set aspect ratio in iOS.

MrHjalmarsson, 2021-03-23:
@cristianoccazinsp sorry, typo in the comment above. I'm of course using screen width for both height and width.

cristianoccazinsp, 2021-03-23:
Ok, I know what's your issue then. It's related to this change
https://github.com/react-native-camera/react-native-camera/pull/2992#issuecomment-802631232

The reason it doesn't happen to me is because I've reverted it on my own
branch, but it should get fixed once
https://github.com/react-native-camera/react-native-camera/pull/3113 is
merged, or the PR reverted.

You can also revert the PR change yourself, or set the pictureSize prop to
"Photo" (iOS only) to force the iOS preset configuration to use the right
one as before the buggy PR.

Let me know if that helps.

MrHjalmarsson, 2021-03-23:
@cristianoccazinsp I'll try that! Thanks for the feedback!
Will let you know

closed 2021-05-31 by stale[bot]