react-native-camera/react-native-camera #3152: Android app stopping after instance of react-native-camera is added

rajwanialiya, 2021-03-22:
# Question

I've added react-native-camera to my application. After creating a component that uses RNCamera, everytime I try to run my app (using `react-native run-android`), it asks for the necessary permissions (camera and audio) and then immediately crashes without outputting any error messages. 

This is my package.json:
```
{
  "main": "index.js",
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "web": "expo start --web",
    "start": "react-native start",
    "test": "jest"
  },
  "dependencies": {
    "@react-native-community/async-storage": "^1.12.1",
    "@tensorflow-models/mobilenet": "^2.0.4",
    "@tensorflow/tfjs": "^3.3.0",
    "@tensorflow/tfjs-react-native": "^0.5.0",
    "expo": "~40.0.0",
    "expo-camera": "~9.1.0",
    "expo-constants": "^10.0.1",
    "expo-gl": "~9.2.0",
    "expo-gl-cpp": "~9.2.0",
    "expo-splash-screen": "~0.8.0",
    "expo-status-bar": "~1.0.3",
    "expo-updates": "~0.4.0",
    "mobx": "^6.1.8",
    "react": "16.13.1",
    "react-dom": "16.13.1",
    "react-native": "~0.63.4",
    "react-native-camera": "^3.43.0",
    "react-native-fs": "^2.16.6",
    "react-native-gesture-handler": "~1.8.0",
    "react-native-reanimated": "~1.13.0",
    "react-native-screens": "~2.15.0",
    "react-native-unimodules": "~0.12.0",
    "react-native-web": "~0.13.12"
  },
  "devDependencies": {
    "@babel/core": "~7.9.0",
    "babel-jest": "~25.2.6",
    "jest": "~25.2.6",
    "react-test-renderer": "~16.13.1"
  },
  "jest": {
    "preset": "react-native"
  },
  "private": true
}
```

This is my build.gradle:
```
// Top-level build file where you can add configuration options common to all sub-projects/modules.
buildscript {
    ext {
        buildToolsVersion = "29.0.2"
        minSdkVersion = 21
        compileSdkVersion = 29
        targetSdkVersion = 29
    }
    repositories {
        google()
        jcenter()
    }
    dependencies {
        classpath("com.android.tools.build:gradle:3.5.3")
        classpath("com.google.gms:google-services:4.3.5")
[... 184 lines cut ...]
  <uses-permission android:name="android.permission.READ_EXTERNAL_STORAGE"/>
  <uses-permission android:name="android.permission.WRITE_EXTERNAL_STORAGE"/>
  <uses-permission android:name="android.permission.WRITE_SETTINGS"/>
  <application android:name=".MainApplication" android:label="@string/app_name" android:icon="@mipmap/ic_launcher" android:roundIcon="@mipmap/ic_launcher_round" android:allowBackup="false" android:theme="@style/AppTheme">
    <meta-data android:name="expo.modules.updates.EXPO_UPDATE_URL" android:value="https://exp.host/@jjwilkin/test_bare"/>
    <meta-data android:name="expo.modules.updates.EXPO_SDK_VERSION" android:value="40.0.0"/>
    <meta-data android:name="expo.modules.updates.ENABLED" android:value="true"/>
    <meta-data android:name="expo.modules.updates.EXPO_UPDATES_CHECK_ON_LAUNCH" android:value="ALWAYS"/>
    <meta-data android:name="expo.modules.updates.EXPO_UPDATES_LAUNCH_WAIT_MS" android:value="0"/>
    <meta-data android:name="com.google.firebase.ml.vision.DEPENDENCIES" android:value="ocr, text" />
    <activity android:name=".MainActivity" android:label="@string/app_name" android:configChanges="keyboard|keyboardHidden|orientation|screenSize|uiMode" android:launchMode="singleTask" android:windowSoftInputMode="adjustResize" android:theme="@style/Theme.App.SplashScreen">
      <intent-filter>
        <action android:name="android.intent.action.MAIN"/>
        <category android:name="android.intent.category.LAUNCHER"/>
      </intent-filter>
    </activity>
    <activity android:name="com.facebook.react.devsupport.DevSettingsActivity"/>
  </application>
</manifest>
```
This is what my android project directory looks like: 

[image]

Any info about why this may be happening or how to resolve this issue would be greatly appreciated! Thanks :)

fabriziobertoglio1987, 2021-03-31:
you have `@tensorflow/tfjs-react-native` which does not use `react-native-camera` but `expo-camera`. I think you should use `expo-camera` installed using unimodules but I'm not sure

rajwanialiya, 2021-03-31:
I wanted to use both cameras for different purposes; expo-camera for image classification because it gives access to the camera stream, and react-native-camera for ocr. Is it not possible to use both cameras (in different tabs of the app)?

fabriziobertoglio1987, 2021-04-01:
it is not a good idea. this project was a fork from expo-camera so probably you end up causing issues

rajwanialiya, 2021-04-01:
So the app is crashing because of a dependency issue? Is there an output I should expect to see, because right now when the RNCamera tab is accessed, the app stops without emitting an error

closed 2021-06-07 by stale[bot]

Imran-cse, 2021-08-23:
Facing the same problem. The app is crashing

AhmedRazaVerge, 2021-09-10:
My React App also crash in when TensorCamera initialised.

`import { Camera } from 'expo-camera';
import { cameraWithTensors } from '@tensorflow/tfjs-react-native';
import React from 'react'
import {View,Platform} from 'react-native';
const TensorCamera = cameraWithTensors(Camera);
export class Realtimedemo extends React.Component {
  handleCameraStream(images, updatePreview, gl) {
    console.log('aaa')
    const loop = async () => {
      const nextImageTensor = images.next().value
      //
      // do something with tensor here
      //
      // if autorender is false you need the following two lines.
      // updatePreview();
      // gl.endFrameEXP();
      requestAnimationFrame(loop);
    }
    loop();
  }
  render() {
   // Currently expo does not support automatically determining the
   // resolution of the camera texture used. So it must be determined
   // empirically for the supported devices and preview size.
   let textureDims;
   if (Platform.OS === 'ios') {
    textureDims = {
      height: 1920,
      width: 1080,
    };
   } else {
    textureDims = {
      height: 1200,
      width: 1600,
    };
   }
   return <View>
     <TensorCamera
      // Standard Camera props
      style={{width:200,height:300}}
      type={Camera.Constants.Type.front}
      // Tensor related props
      cameraTextureHeight={textureDims.height}
      cameraTextureWidth={textureDims.width}
      resizeHeight={200}
      resizeWidth={152}
      resizeDepth={3}
      onReady={this.handleCameraStream}
      autorender={true}
     />
   </View>
  }
}`

`import React, {Fragment} from 'react';
import {StyleSheet, View, Text} from 'react-native';
import * as tf from '@tensorflow/tfjs';
import '@tensorflow/tfjs-react-native';
import Webcam_demo from './src/components/webcam/webcam_demo';
import {Realtimedemo} from './src/components/webcam/Realtimedemo'

const BACKEND_TO_USE = 'cpu';
// const BACKEND_TO_USE = 'rn-webgl';
class App extends React.Component {
  state = {
    isTfReady: false,
  };

  async componentDidMount() {
    await tf.setBackend(BACKEND_TO_USE);
    await tf.ready();
    let a= await tf.getBackend()
    this.setState({
      isTfReady: true,
    });

    //Output in Expo console
    console.log(this.state.isTfReady);
  }

  render() {
    return (
      <View style={styles.container}>
        <Text>TFJS ready? {this.state.isTfReady ? <Text>Yes</Text> : ''}</Text>
        {this.state.isTfReady ? <Realtimedemo /> : null}
      </View>
    );
  }
}

const styles = StyleSheet.create({
  container: {
    flex: 1,
    backgroundColor: '#fff',
    alignItems: 'center',
    justifyContent: 'center',
  },
});

export default App;
`

michelazzam, 2022-03-11:
@rajwanialiya How did you manage to solve this issue ?