react-native-camera/react-native-camera #3153: Manual focus depth changes while capturing image

Sandra12-6, 2021-03-22:
**Related Modules**
RNCamera

**Platforms**
Android (redmi note 8 pro)

**Versions**
react-native-camera: 3.42.3

**Description/Current Behaviour**
The camera preview changes and becomes out of focus while capturing the image, resulting in an image different from the preview before capture.

**Expected Behaviour**
The focus/blur level should not change after capturing an image

**Here is my code**

```
<RNCamera
ref={(ref) => {
setCameraRef(ref);
}}
ratio="9:16"
style={styles.preview}
captureAudio={false}
focusDepth={1}
autoFocus={RNCamera.Constants.AutoFocus.off}
useCamera2Api={true}
type={RNCamera.Constants.Type.back}
flashMode={RNCamera.Constants.FlashMode.off}
whiteBalance={RNCamera.Constants.WhiteBalance.shadow}
androidCameraPermissionOptions={{
title: 'Permission to use camera',
message: 'We need your permission to use your camera',
buttonPositive: 'Ok',
buttonNegative: 'Cancel',
}}>

<TouchableOpacity 
	  onPress={async() => {
if (cameraRef) {
  const options = { quality: 1, base64: true };
  const data = await cameraRef.takePictureAsync(options);
  navigation.navigate('Image',{'photo':data.uri});
      }),
    });
}
	  }}
style={styles.capture}>
<Text style={{ fontSize: 14, color:'white', fontWeight:'bold'}}> TAKE PICTURE </Text>
</TouchableOpacity>
	</View>
</RNCamera>
```
[image]
[image]

@cristianoccazinsp @abrahambotros @lwansbrough @MarcoScabbiolo  thoughts? any help is greatly appreciated!

cristianoccazinsp, 2021-03-23:
Try without `useCamera2Api`, Camera2 is quite broken. Do you need any specific feature from it?

Also, try without `focusDepth={1}`, is there any specific reason you need this?

Sandra12-6, 2021-03-23:
@cristianoccazinsp I'm using Camera2 API because the focusDepth property doesn't work with Camera1.
I eventually want to have a slider to change the focusDepth (code below). I tested it on Galaxy A6 phone, it worked well, the focus/blur level didn't change after capturing an image. but on the redmi phone it does change,... 

```
  const [focusValue, setFocusValue] = useState(1);
  <RNCamera
     ref={(ref) => {
     setCameraRef(ref);
          }}
      focusDepth={focusValue}
        ........
		
    <Slider
    minimumValue={0}
    maximumValue={1}
    allowTouchTrack={true}
    animateTransitions={true}
    minimumTrackTintColor="white"
    maximumTrackTintColor="#15a2ea"
    thumbTintColor="#bae1f5"
    style={{marginLeft:20, marginRight:20}}
    thumbStyle={{width:20, height:20}}
    value={0}
    onValueChange={(sliderValue) => setFocusValue(sliderValue)}
  /> 
```

cristianoccazinsp, 2021-03-23:
I can't really help with camera2 as I always avoid it due to all the bugs
it has.  I remember having a similar issue where the preview would focus
and then when capturing, the camera would trigger another focus attempt and
end up taking a couple of seconds to capture, as opposed to camera1
capturing instantly.

Sandra12-6, 2021-03-23:
@cristianoccazinsp I see, do you know any way I can add manual focus to a camera then? or my only option is the autoFocus?

cristianoccazinsp, 2021-03-23:
Not that I'm aware of. Either camera2 is fixed, or you resort to auto focus
and tao to focus at least.

Keep in mind there are some devices that will still fallback to camera1,
even if they use an updated android version. So chances are you will also
have compatibility issues with varios devices if you rely on camera2
features.

You may also check https://github.com/cuvent/react-native-vision-camera to
see if it works for your use case. I believe this may be the next version
for react native camera.

Sandra12-6, 2021-03-23:
@cristianoccazinsp I will check it! Thank you very much really :))

fabriziobertoglio1987, 2021-03-31:
https://github.com/react-native-camera/react-native-camera/blob/ed84c27ff840f3d499377645b4e314c43cd39059/android/src/main/java/com/google/android/cameraview/Camera2.java#L638-L654

https://github.com/react-native-camera/react-native-camera/blob/c591e9d4e07fdd973e8c1cf681d2c5e10025aa0f/android/src/main/java/com/google/android/cameraview/Camera2.java#L1116-L1129

Sandra12-6, 2021-03-31:
Hi @fabriziobertoglio1987 this is how I have it in Camera2.java file, do I need to change anything else?

fabriziobertoglio1987, 2021-03-31:
@Sandra12-6  you need to build your project with android studio and search for that file 

Debug this line
```java
mPreviewRequestBuilder.set(CaptureRequest.LENS_FOCUS_DISTANCE, value);
```
https://developer.android.com/studio/debug

once you did that, you can send me your feedback and I will help you forward

otherwise I'll be able to work on this only when I'll have free time .. not soon. Thanks :pray: :peace_symbol:

Sandra12-6, 2021-03-31:
Ok I will, thank you! @fabriziobertoglio1987

closed 2021-06-07 by stale[bot]