react-native-camera/react-native-camera #3154: Cannot find symbol retriever.close()  [Urgent, Solution included]

railsjack, 2021-03-25:
# Bug Report

**Platforms**
Android

**Versions**
3.43.0

- Android: 28
- react-native-camera: 3.43
- react-native: 0.62.0
- react: 16.13.1
- react-navigation: 4.1.1

**Description/Current Behaviour**

When I try to compile the react-native project, it fails.

**Expected Behaviour**

Compilation & Installation on Emulator should work.

**Steps to Reproduce**

It's simple. 
`react-native run-android`

**Additionals**
[image]

**Reason and Solution for this issue**
I think that it's because there is no method named "close" in the class `android.media.MediaMetaDataRetriever` on Android API 28.
https://github.com/react-native-camera/react-native-camera/blob/8aad81d6ebd42c73b70cbb45784d69f24db80ea9/android/src/main/java/org/reactnative/camera/CameraModule.java#L536
As @cristianoccazinsp mentioned here, it does not work on API 28.
By the way, my app should work since API 28 at least.
So, I want you to put a solution here using Java Reflection.
Here is a good example of it.
https://stackoverflow.com/questions/160970/how-do-i-invoke-a-java-method-when-given-the-method-name-as-a-string

QuickChangeSpidey, 2021-03-26:
Change method on native android module cameraModule.java to receiver.release(). Rebuild on Android studio. I believe this needs to be fixed in the version update.

akshay247-voiceit, 2021-03-26:
Downgrade your react native camera version to: "react-native-camera": "3.42.3", in package.json and this does not occur. I think they broke something in the latest release for API<29

QuickChangeSpidey, 2021-03-26:
Downgrade your react native camera version to: "react-native-camera": "3.42.3", in package.json and this does not occur. I think they broke something in the latest release for API<29

railsjack, 2021-03-26:
Yeah, I also think downgrading of the version is good for now.

cristianoccazinsp, 2021-03-26:
@QuickChangeSpidey does `retriever.release()` behave just like close? The goal is to be able to use this class without leaking memory or resources.

Although I still think SDK 29 should be eventually used (RN 0.64 is using it already, and google will no longer accept apps compiled with 28), it's probably a good change to replace `close` with `release` as long as it behaves the same.

railsjack, 2021-03-26:
@cristianoccazinsp , Yeah. As a native developer, I think it seems a better choice for you to use "retriever.release()" than "retriever.close()".

closed 2021-03-26 by railsjack

n1ru4l, 2021-03-26:
:tada: This issue has been resolved in version 3.43.1 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera/v/3.43.1)
- [GitHub release](https://github.com/react-native-camera/react-native-camera/releases/tag/v3.43.1)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket: