react-native-camera/react-native-camera #3160: useNativeZoom sensitivity

msageryd, 2021-03-30:
useNativeZoom is a great feature. I could get rid of a lot of pinch-zoom code and instead just set `useNativeZoom=true`.

Unfortunately nativeZoom is very sensitive. The zoom level increases way faster than the actual pinch. It feels as if the zoom is exponential in relation to the pinch delta. My users thinks the zoom is "yanky" compared to my old pinch-zoom code. I'd rather have less code, but I also need happy users.

Is there a property for altering the zoom speed?

fabriziobertoglio1987, 2021-03-31:
https://github.com/react-native-camera/react-native-camera/blob/42388be530f4a791ff6822b430b346248c2a40af/android/src/main/java/org/reactnative/camera/RNCameraView.java#L422-L429

msageryd, 2021-03-31:
Thanks, but I don't know what to do with this. Is this a suggested PR?
Anyway, my problem is with iOS. Haven't tried useNativeZoom on Android yet.

fabriziobertoglio1987, 2021-04-01:
https://github.com/react-native-camera/react-native-camera/blob/c591e9d4e07fdd973e8c1cf681d2c5e10025aa0f/ios/RN/RNCamera.m#L1739-L1748

I don't do iOS development but If the solution comes to my mind I'll publish the pr

closed 2021-06-09 by stale[bot]