react-native-camera/react-native-camera #3164: on rotaing App continusely the camera is stuck at landscape mode

AjayShivanagol, 2021-03-30:
[image]

fabriziobertoglio1987, 2021-03-31:
how do we know this is not emulator issue?

AjayShivanagol, 2021-03-31:
@fabriziobertoglio1987 you are right it's an emulator issue I tried on a real device it's working fine.

closed 2021-03-31 by AjayShivanagol

fabriziobertoglio1987, 2021-03-31:
@AjayShivanagol thanks a lot. Yes, it is unlucky but we have many issues that affect our ability to create React Native Apps, so we need to prioritize. We will allways have limited number of developers doing open source.

I believe we should have libraries that have no bugs/issues so that developers can be productive while building apps with reactnative. :peace_symbol: :pray: 

I believe the community should decide which issues needs to be prioritized. I just comment so I'll remember next time what was my last opinion on the matter.