react-native-camera/react-native-camera #3166: CocoaPods could not find compatible versions for pod "Firebase/CoreOnly"

anoopjangra90, 2021-03-31:
while installing Pod , I'm facing following error :

 In Podfile:
    RNFBApp (from `../node_modules/@react-native-firebase/app`) was resolved to 11.2.0, which depends on
      Firebase/CoreOnly (= 7.9.0)

    react-native-camera/FaceDetectorMLKit (from `../node_modules/react-native-camera`) was resolved to 3.43.1, which depends on
      Firebase/MLVisionFaceModel was resolved to 5.0.0, which depends on
        Firebase/CoreOnly (= 5.0.0)

Packages 👍 

"react-native-camera": "^3.43.1",
"@react-native-firebase/app": "^11.2.0",

sadruddyn, 2021-03-31:
same .. but mine is for Firebase/MLVisionTextModel  :/

anoopjangra90, 2021-03-31:
@sadruddyn [Check this](https://github.com/react-native-camera/react-native-camera/issues/3043)

sadruddyn, 2021-03-31:
error resolved but not working .. what about you?

anoopjangra90, 2021-04-01:
same here @sadruddyn

manjinderr, 2021-04-27:
Any update when this issue would be fixed.

I also see similar error:
[!] CocoaPods could not find compatible versions for pod "Firebase/CoreOnly":
 In Podfile:
  RNFBApp (from `../node_modules/@react-native-firebase/app`) was resolved to 11.3.3, which depends on
   Firebase/CoreOnly (= 7.10.0)
  react-native-camera/TextDetector (from `../node_modules/react-native-camera`) was resolved to 3.43.6, which depends on
   Firebase/MLVisionTextModel was resolved to 5.0.0, which depends on
    Firebase/CoreOnly (= 5.0.0)

Packages used:
"react-native-camera": "^3.43.6",
"@react-native-firebase/app": "^11.3.3",

nthegedus, 2021-05-13:
Same issue here

```
"react-native-camera": "3.43.6",
"react-native-firebase/app": "^11.3.3"
```

wouterds, 2021-05-27:
So the only solution here is downgrading react-native-firebase to somehow try to magically match with the version required by this lib? 💩

nthegedus, 2021-06-01:
@wouterds 

I think that's the temporary solution until the v4 be released
https://github.com/react-native-camera/react-native-camera/issues/3100

The highest firebase libs version I could use without any error are:
```
"@react-native-firebase/app": "^8.4.7",
"@react-native-firebase/analytics": "^10.0.0",
"@react-native-firebase/crashlytics": "^8.5.2",
"@react-native-firebase/dynamic-links": "^7.5.13",
"@react-native-firebase/messaging": "^8.0.1",
"@react-native-firebase/remote-config": "^10.0.1",
```

nthegedus, 2021-06-01:
@wouterds 

If it's your concern. Using those Firebase lib versions will not be reject by Apple by **App Tracking Transparency** Rules, only if your app is **NOT** made for kids. Otherwise you'll have a big trouble.

wouterds, 2021-06-01:
It's not a concern, I am using Firebase libs and they're conflicting with this one. You can opt-out entirely from tracking btw.

vemulanishanth, 2021-06-14:
@nthegedus
`
npm ERR! code ERESOLVE
npm ERR! ERESOLVE unable to resolve dependency tree
npm ERR! 
npm ERR! While resolving: reactnativecore@0.0.1
npm ERR! Found: @react-native-firebase/app@8.4.7
npm ERR! node_modules/@react-native-firebase/app
npm ERR!   @react-native-firebase/app@"^8.4.7" from the root project
npm ERR! 
npm ERR! Could not resolve dependency:
npm ERR! peer @react-native-firebase/app@"9.0.0" from @react-native-firebase/messaging@8.0.1
npm ERR! node_modules/@react-native-firebase/messaging
npm ERR!   @react-native-firebase/messaging@"^8.0.1" from the root project
npm ERR! 
npm ERR! Fix the upstream dependency conflict, or retry
npm ERR! this command with --force, or --legacy-peer-deps
npm ERR! to accept an incorrect (and potentially broken) dependency resolution.
`
can you share the exact versions of the firebase 'app and messaging' that worked for you?

nthegedus, 2021-06-15:
@vemulanishanth here, how is on my package.json

```
"@react-native-firebase/analytics": "^10.0.0",
"@react-native-firebase/app": "^8.4.7",
"@react-native-firebase/crashlytics": "^8.5.2",
"@react-native-firebase/dynamic-links": "^7.5.13",
"@react-native-firebase/messaging": "^8.0.1",
"@react-native-firebase/remote-config": "^10.0.1",
```

vemulanishanth, 2021-06-16:
> @vemulanishanth here, how is on my package.json
>
> ```
> "@react-native-firebase/analytics": "^10.0.0",
> "@react-native-firebase/app": "^8.4.7",
> "@react-native-firebase/crashlytics": "^8.5.2", [...]

Thanks for the reply. I couldn't make it work though. I just need one of the functionalities this package supports.. I'm planning on replacing the package altogether with any other package or write one myself for my specific purpose.

closed 2021-07-15 by MateusAndrade

MateusAndrade, 2021-07-16:
Released at: [v4.0.0](https://github.com/react-native-camera/react-native-camera/releases/tag/v4.0.0)

DevThou, 2022-01-29:
I fixed this by changing firebase/app version to 
  "@react-native-firebase/app": "^10.8.0".

Good luck