react-native-camera/react-native-camera #3170: Stutters on video record start

eqlion, 2021-04-01:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: 10
- iOS: — 
- react-native-camera: 3.38.0
- react-native: 0.62.2
- react: 16.11.0
  

**Description/Current Behaviour**

When starting recording video, there's a visible stutter, only after which the actual recording starts

https://user-images.githubusercontent.com/18642157/113299825-f46f8180-9305-11eb-9507-b500f5fd6765.mp4

**Expected Behaviour**

No such stutter, immediate start of recording 

**Steps to Reproduce**

```tsx
export default () => {
  const ref = useRef<RNCamera>(null);
  return (<RNCamera ref={ref} style={styles.root} type={cameraType}>
    <Button onPress={()=>ref.current.recordAsync({fps: 30, quality: RNCamera.Constants.VideoQuality["720p"], maxDuration: 60000})
  </RNCamera>)}
```

**Additionals**

> Love react-native-camera? Please consider supporting our collective: 👉 https://opencollective.com/react-native-camera/donate
> Want this issue to be resolved faster? Please consider adding a bounty to it https://issuehunt.io/repos/33218414

cristianoccazinsp, 2021-04-01:
On iOS, the stutter happens because the camera has to switch from "Photo" to "Video" preset. This is a major drawback for using photo and video simultaneously on the same component (as opposed to the native camera which switches between modes).

If you really want to prevent this, try setting the prop (for IOS only) `pictureSize="1280x720"`.  Doing so, the default camera preset will match your video recording configuration, and so the stutter should go away. Note that picture quality will be affected if you take photos with this preset.

Alternatively, listen to the `onRecordingStart` and `onRecordingEnd` events to better adjust your UI to what's really happening.

eqlion, 2021-04-01:
@cristianoccazinsp yes, I've seen such issues and tried adding pictureSize. However, the problem occured on Android  and it wasn't resolved through setting this prop, unfortunately.

cristianoccazinsp, 2021-04-01:
On Android, I don't think there's a lot you can do. You most likely need to listen to the `onRecordingStart` and handle the stutter through the UI. The native app blurs the screen briefly when switching from one mode to another, and so on.

closed 2021-06-09 by stale[bot]