react-native-camera/react-native-camera #3178: Android app crashes because of Exceptions from BarCodeScannerAsyncTask

jeremydeal, 2021-04-10:
# Bug Report

**To Do First**

- [X] Did you try latest release?
- [X] Did you try master?
- [X] Did you look for existing matching issues?

**Platforms**

Android

**Versions**

- Android: All versions. See root cause below.

**Description/Current Behaviour**

Our logs show about a dozen Android app crashes per day while users attempt to scan QR codes. We haven't seen any iOS crashes.

Here's a representative stack trace from app crash (logged in Sentry):

```
java.lang.RuntimeException: 
  at android.os.AsyncTask$4.done (AsyncTask.java:399)
  at java.util.concurrent.FutureTask.finishCompletion (FutureTask.java:383)
  at java.util.concurrent.FutureTask.setException (FutureTask.java:252)
  at java.util.concurrent.FutureTask.run (FutureTask.java:271)
  at android.os.AsyncTask$SerialExecutor$1.run (AsyncTask.java:289)
  at java.util.concurrent.ThreadPoolExecutor.runWorker (ThreadPoolExecutor.java:1167)
  at java.util.concurrent.ThreadPoolExecutor$Worker.run (ThreadPoolExecutor.java:641)
  at java.lang.Thread.run (Thread.java:919)
Caused by: java.lang.ArrayIndexOutOfBoundsException: 
  at com.google.zxing.pdf417.decoder.ec.ModulusGF.multiply (ModulusGF.java:105)
  at com.google.zxing.pdf417.decoder.ec.ModulusPoly.evaluateAt (ModulusPoly.java:99)
  at com.google.zxing.pdf417.decoder.ec.ErrorCorrection.decode (ErrorCorrection.java:53)
  at com.google.zxing.pdf417.decoder.PDF417ScanningDecoder.correctErrors (PDF417ScanningDecoder.java:557)
  at com.google.zxing.pdf417.decoder.PDF417ScanningDecoder.decodeCodewords (PDF417ScanningDecoder.java:530)
  at com.google.zxing.pdf417.decoder.PDF417ScanningDecoder.createDecoderResultFromAmbiguousValues (PDF417ScanningDecoder.java:309)
  at com.google.zxing.pdf417.decoder.PDF417ScanningDecoder.createDecoderResult (PDF417ScanningDecoder.java:278)
  at com.google.zxing.pdf417.decoder.PDF417ScanningDecoder.decode (PDF417ScanningDecoder.java:124)
  at com.google.zxing.pdf417.PDF417Reader.decode (PDF417Reader.java:87)
  at com.google.zxing.pdf417.PDF417Reader.decode (PDF417Reader.java:61)
  at com.google.zxing.MultiFormatReader.decodeInternal (MultiFormatReader.java:171)
  at com.google.zxing.MultiFormatReader.decodeWithState (MultiFormatReader.java:85)
  at org.reactnative.camera.tasks.BarCodeScannerAsyncTask.doInBackground (BarCodeScannerAsyncTask.java:100)
  at org.reactnative.camera.tasks.BarCodeScannerAsyncTask.doInBackground (BarCodeScannerAsyncTask.java:10)
  at android.os.AsyncTask$3.call (AsyncTask.java:378)
  at java.util.concurrent.FutureTask.run (FutureTask.java:266)
```

**Expected Behaviour**

I expect BarCodeScannerAsyncTask.doInBackground() to catch the error and simply return null, which is clearly the intended behavior, given the `(Throwable t)` on line 132.

**Steps to Reproduce**

I have an MVP in a Java unit test, pasted below.

In `theirCode()`, I copy-pasted the code from `BarCodeScannerAsyncTask.doInBackground()`, then replaced the code inside each **try** with a function that would reliably throw exceptions. I was able to demonstrate that an **ArrayIndexOutOfBoundsException** would indeed bubble up out of the third **try**.

In `myCode()`, I restructured the method in a way that preserves the same logic but does not allow exceptions to bubble up past that last catch. I will use this structure as the basis for a PR and submit the fix soon.

```
package com.jdeal;

import static org.junit.jupiter.api.Assertions.*;
import org.junit.jupiter.api.Test;

class MainTest {

    @Test
    void theirCode() {
        new TestReactNativeCameraCatches().theirCode();
    }

    @Test
    void myCode() {
        new TestReactNativeCameraCatches().myCode();
    }
}

class TestReactNativeCameraCatches {

    public String theirCode() {
        try {
            throwNullPointerException();
        } catch (NullPointerException e) {
            try {
                throwNullPointerException();
            } catch (NullPointerException e1) {
                try {
                    String hi = "hi";
                    throw new ArrayIndexOutOfBoundsException("Just testing.");
                } catch (NullPointerException e2) {
                    try {
                        throwNullPointerException();
                    } catch (NullPointerException e3) {
                        // nothing
                    }
                }
            }
        } catch (Throwable t) {
            //t.printStackTrace();
        }

        return "hi!";
    }

    public String myCode() {
        String output = null;

        try {
            try {
                output = throwNullPointerException();
                return output;
            } catch (NullPointerException e) {
            }

            try {
                output = throwNullPointerException();
                return output;
            } catch (NullPointerException e1) {
            }

            try {
                String hi = "hi";
                throw new ArrayIndexOutOfBoundsException("Just testing.");
            } catch (NullPointerException e2) {
            }

            try {
                throwNullPointerException();
            } catch (NullPointerException e3) {
                // nothing
            }
        } catch (Throwable t) {
            //t.printStackTrace();
        }

        return "hi!";
    }

    private String throwNullPointerException() {
        String s = null;
        return s.toString();
    }
}
```

**Additionals**

Thanks for the work on this library! 😄

jeremydeal, 2021-04-11:
PR created: https://github.com/react-native-camera/react-native-camera/pull/3179

closed 2021-04-11 by MateusAndrade

n1ru4l, 2021-04-11:
:tada: This issue has been resolved in version 3.43.5 :tada:

The release is available on:
- [npm package (@latest dist-tag)](https://www.npmjs.com/package/react-native-camera/v/3.43.5)
- [GitHub release](https://github.com/react-native-camera/react-native-camera/releases/tag/v3.43.5)

Your **[semantic-release](https://github.com/semantic-release/semantic-release)** bot :package::rocket: