react-native-camera/react-native-camera #3187: GoogleVisionBarcodes is not working in iOS

Haseeba393, 2021-04-15:
# Question
I have checked all the documentation of react-native-camera. I'm using onGoogleVisionBarcodesDetected for barcode deetction. It is working great for Android and when i'm opening my app in iphone, it is crashing while coming on camera screen. 

# System Info
System:
    OS: macOS 10.15.6
    CPU: (4) x64 Intel(R) Core(TM) i5-4308U CPU @ 2.80GHz
    Memory: 211.28 MB / 8.00 GB
    Shell: 5.7.1 - /bin/zsh
  Binaries:
    Node: 12.18.3 - /usr/local/bin/node
    Yarn: 1.22.10 - /usr/local/bin/yarn
    npm: 7.6.3 - /usr/local/bin/npm
    Watchman: 4.9.0 - /usr/local/bin/watchman
  Managers:
    CocoaPods: 1.10.1 - /usr/local/bin/pod
  SDKs:
    iOS SDK:
      Platforms: iOS 13.6, DriverKit 19.0, macOS 10.15, tvOS 13.4, watchOS 6.2
    Android SDK: Not Found
  IDEs:
    Android Studio: 4.1 AI-201.8743.12.41.7199119
    Xcode: 11.6/11E708 - /usr/bin/xcodebuild
  Languages:
    Java: javac 15 - /usr/bin/javac
    Python: 2.7.16 - /usr/bin/python
  npmPackages:
    @react-native-community/cli: Not Found
    react: 16.13.1 => 16.13.1 
    react-native: 0.63.4 => 0.63.4 
    react-native-macos: Not Found
  npmGlobalPackages:
    *react-native*: Not Found

# Podfile
require_relative '../node_modules/react-native/scripts/react_native_pods'
require_relative '../node_modules/@react-native-community/cli-platform-ios/native_modules'

platform :ios, '10.0'

target 'pr' do
  config = use_native_modules!

  use_react_native!(:path => config["reactNativePath"])

  # Third party deps podspec link
  # pod 'GoogleAppMeasurement', '~> 5.3.0'
  # pod 'nanopb', '~> 0.3'
  pod 'react-native-camera', path: '../node_modules/react-native-camera', subspecs: [
    'BarcodeDetectorMLKit'
  ]

  target 'prTests' do
    inherit! :complete
    # Pods for testing
  end

  # post_install do |installer|
  #   installer.pods_project.targets.each do |target|
  #     if target.name == "React"
  #       target.remove_from_project
  #     end
  #   end
  # end

  # Enables Flipper.
  #
  # Note that if you have use_frameworks! enabled, Flipper will not work and
  # you should disable these next few lines.
  # use_flipper!
  # post_install do |installer|
  #   flipper_post_install(installer)
  # end
end

target 'pr-tvOS' do
  # Pods for pr-tvOS

  target 'pr-tvOSTests' do
    inherit! :search_paths
    # Pods for testing
  end
end

# Package.json
{
  "name": "pr",
  "version": "0.0.1",
  "private": true,
  "scripts": {
    "android": "react-native run-android",
    "ios": "react-native run-ios",
    "start": "react-native start",
    "test": "jest",
    "lint": "eslint ."
  },
  "dependencies": {
    "@gorhom/animated-tabbar": "^2.1.0",
    "@react-native-async-storage/async-storage": "^1.13.4",
    "@react-native-community/checkbox": "^0.5.7",
    "@react-native-community/datetimepicker": "^3.0.9",
    "@react-native-community/masked-view": "^0.1.10",
    "@react-native-community/progress-bar-android": "^1.0.4",
    "@react-native-community/progress-view": "^1.2.4",
    "@react-native-community/push-notification-ios": "^1.8.0",
    "@react-native-firebase/app": "^11.1.0",
    "@react-native-firebase/messaging": "^11.1.0",
    "@react-navigation/bottom-tabs": "^5.11.2",
    "@react-navigation/drawer": "^5.12.3",
    "@react-navigation/material-top-tabs": "^5.3.13",
    "@react-navigation/native": "^5.8.10",
    "@react-navigation/stack": "^5.12.8",
    "axios": "^0.21.1",
    "jwt-decode": "^3.1.2",
    "react": "16.13.1",
    "react-native": "0.63.4",
    "react-native-action-button": "^2.8.5",
    "react-native-axios": "^0.17.1",
    "react-native-camera": "3.40.0",
    "react-native-chart-kit": "^6.8.1",
    "react-native-countdown-circle-timer": "^2.5.0",
    "react-native-device-info": "^8.0.5",
    "react-native-document-picker": "^4.2.0",
    "react-native-file-viewer": "^2.1.4",
    "react-native-fs": "^2.13.2",
    "react-native-gesture-handler": "^1.9.0",
    "react-native-image-base64": "^0.1.4",
    "react-native-image-picker": "^3.1.4",
    "react-native-material-dropdown": "github:noway/react-native-material-dropdown",
    "react-native-modal": "^11.6.1",
    "react-native-modal-datetime-picker": "^9.1.0",
    "react-native-otp-inputs": "^6.0.0",
    "react-native-pdf": "^6.2.2",
    "react-native-popup-menu": "^0.15.10",
    "react-native-push-notification": "^7.2.2",
    "react-native-reanimated": "^1.13.2",
    "react-native-responsive-screen": "^1.4.2",
    "react-native-safe-area-context": "^3.1.9",
    "react-native-screens": "^2.16.1",
    "react-native-share": "^5.1.1",
    "react-native-step-indicator": "^1.0.3",
    "react-native-svg": "^12.1.0",
    "react-native-swipeable": "^0.6.0",
    "react-native-tab-view": "^2.15.2",
    "react-native-vector-icons": "^7.1.0",
    "react-native-webview": "^11.2.3",
    "react-redux": "^7.2.2",
    "redux": "^4.0.5",
    "redux-thunk": "^2.3.0",
    "rn-fetch-blob": "^0.12.0",
    "rn-swipe-button": "^1.3.3"
  },
  "devDependencies": {
    "@babel/core": "7.12.10",
    "@babel/runtime": "7.12.5",
    "@react-native-community/eslint-config": "1.1.0",
    "babel-jest": "25.5.1",
    "eslint": "6.8.0",
    "jest": "25.5.4",
    "metro-react-native-babel-preset": "0.59.0",
    "react-test-renderer": "16.13.1"
  },
  "jest": {
    "preset": "react-native"
  }
}

# Error I'm getting while doing pod install
[!] CocoaPods could not find compatible versions for pod "Firebase/CoreOnly":
  In snapshot (Podfile.lock):
    Firebase/CoreOnly (= 7.8.0)

  In Podfile:
    RNFBApp (from `../node_modules/@react-native-firebase/app`) was resolved to 11.1.0, which depends on
      Firebase/CoreOnly (= 7.8.0)

    react-native-camera/BarcodeDetectorMLKit (from `../node_modules/react-native-camera`) was resolved to 3.40.0, which depends on
      Firebase/MLVisionBarcodeModel was resolved to 5.0.0, which depends on
        Firebase/CoreOnly (= 5.0.0)

You have either:
 * out-of-date source repos which you can update with `pod repo update` or with `pod install --repo-update`.
 * changed the constraints of dependency `Firebase/CoreOnly` inside your development pod `react-native-camera/BarcodeDetectorMLKit`.
   You should run `pod update Firebase/CoreOnly` to apply changes you've made.

GunnarAK, 2021-04-23:
Obviously if you get pod errors, trying to start the app won't suddenly have you succeed.

Maybe my setup can help you get past the pod error: https://github.com/react-native-camera/react-native-camera/issues/3040#issuecomment-825635591

Haseeba393, 2021-04-23:
@GunnarAK . Your setup only will not work. In actual, podspec file in react-native-camera/ios is outdated in newer versino of react-native-camera.
Some of lines has to be updated in podspec file to get rid to this error. I did those changes and now app is running fine but barcode scanning is not working on only iOS side

closed 2021-08-23 by stale[bot]