react-native-camera/react-native-camera #3197: Writing UserComment exif data not working?  Or lack of documentation?

bneigher, 2021-04-30:
# Bug Report

**To Do First**

- [x] Did you try latest release?
- [x] Did you try master?
- [x] Did you look for existing matching issues?

**Platforms**
All

**Versions**
- iOS: 14.5
- react-native-camera: 3.43.6 
- react-native: 0.64.0
- react: 17.0.1

**Description/Current Behaviour**
I am trying to store some information into the UserComment exif field.  It seems that no matter what I try "string", "object", "object with stringified properties", the results is always an empty json object (ASCII{}) when viewing in exiftool cli.

```
[~/Desktop/Code]$ exiftool /Users/ben/Desktop/img.jpg  | grep User\ Comment
User Comment                    : {}
```

I have seen some examples of people writing GPS exif values with react native camera, and I can see that it is trying to write to UserComment because without my code attempts the property doesn't showup when viewing the exif data.

**Expected Behaviour**
I expect the output of [exiftool](https://exiftool.org/) to show a stringified object when reading a picture.

```
[~/Desktop/Code]$ exiftool /Users/ben/Desktop/img.jpg  | grep User\ Comment
User Comment                    : {"myprop": "myvalue"}
```

**Steps to Reproduce**

```javascript
const options = {
  exif: true,
  writeExif: {
    UserComment: JSON.stringify({
      "myprop": "myvalue"
    }),
  },
  fixOrientation: true,
  forceUpOrientation: true,
  skipProcessing: true,
}
const picture = await this.camera.current.takePictureAsync(options);
```

bneigher, 2021-04-30:
Or if anything if someone can explain how this works in swift with json:
https://github.com/react-native-camera/react-native-camera/blob/941536592cb8ad6f65f3cb50e2fe814b7c70c093/ios/RN/RNCamera.m#L882

closed 2021-07-06 by stale[bot]

Pakile, 2022-03-31:
Hi @bneigher 
You have any solutions?
I have the same issue for iOS :(
Thanks