react-native-camera/react-native-camera #3227: Please update FirebaseMLVisionBarcodeModel ➡️ MLKitBarcodeScanning to remove warnings

adrian-tabirta, 2021-05-12:
Please update FirebaseMLVisionBarcodeModel ➡️ MLKitBarcodeScanning to remove warnings

have:`"react-native-camera": "^3.43.6",` and on: `pod install` I got: 
```
[!] FirebaseMLCommon has been deprecated

[!] FirebaseMLVision has been deprecated

[!] FirebaseMLVisionBarcodeModel has been deprecated in favor of MLKitBarcodeScanning
```

MateusAndrade, 2021-05-12:
Hey @adrian-tabirta , I think we will need to follow the migration guide descried here:

- Android: https://developers.google.com/ml-kit/migration/android
- iOS: https://developers.google.com/ml-kit/migration/ios

Btw, we need to read about in order to determine if we will need to change something in the native code, which can lead to some breaking changes.

cristianoccazinsp, 2021-05-18:
Do we really need Firebase/MLKit to do barcode scanning? I have been using the default implementation for months and it works wonders for both Barcodes and QR codes.

prettydev, 2021-05-20:
@MateusAndrade , Can you explain more details on how to migrate with some example code and project directory?
I checked both migration guides((https://developers.google.com/ml-kit/migration/android, https://developers.google.com/ml-kit/migration/ios)) you mentioned, but not sure and confusing how to apply this guide to the ReactNative project.
I'm usually working with ReactNative, but for the native parts for Android and IOS, I'm still unfamiliar, so can't find the "old" code blocks mentioned on the guide. I want to know how to combine ReactNative code and native code and keep the project ongoing.

cristianoccazinsp, 2021-05-20:
The library already has native implementations for iOS and Android, I have no idea why there's also a MLKit version for it, but I guess it adds support to a few more codes.

Bill-Niz, 2021-06-07:
> Do we really need Firebase/MLKit to do barcode scanning? I have been using the default implementation for months and it works wonders for both Barcodes and QR codes.

how do you use the default ones with RNCamera?

cristianoccazinsp, 2021-06-07:
I have started to get the following warning even though I don't use MLKit at all from face detection.

```
WARNING: [Processor] Library '.gradle/caches/modules-2/files-2.1/com.google.firebase/firebase-ml-vision/19.0.3/53dffc1401dc7d4e5e6c87a6de7ce0984d1274b1/firebase-ml-vision-19.0.3.aar' contains references to both AndroidX and old support library. This seems like the library is partially migrated. Jetifier will try to rewrite the library anyway.
 Example of androidX reference: 'androidx/exifinterface/media/ExifInterface'
```

What's worse, trying to set the firebase-ml-vision variable names as seen in the build config is not possible due to the hyphens: This doesn't work.
```
firebase-ml-vision = '24.1.0'               // react-native-camera, unused
firebase-ml-vision-face-model = '20.0.2'    // react-native-camera, unused
```

Variable names were taken from here: https://github.com/react-native-camera/react-native-camera/blob/master/android/build.gradle#L79

Is it possible to not get any download/usage of the mlkit lib if it is not being used? Even with `missingDimensionStrategy 'react-native-camera', 'general'`, looks like the mlkit stuff is getting downloaded/compiled.

lfrkovic-spaceiq, 2021-07-09:
> The library already has native implementations for iOS and Android, I have no idea why there's also a MLKit version for it, but I guess it adds support to a few more codes.

How do you use native implementations in react-native-camera?

closed 2021-07-15 by MateusAndrade